openWB/core #229: Fehler in JSON Modulen

DetMoerk, 2021-12-28:
Die Ermittlung der Werte mit den JQ Filtern funktioniert, allerdings wird ein Fehler beim Payload geworfen, was vermutlich dazu führt, dass die MQTT Topics nicht gefüllt werden. Die Zählerstände für EVU und PV werden korrekt ausgelesen. Bei EVU wird auch die aktuelle Leistung im Status und im Graphen angezeigt. 
Bei PV wird der Wert der Leistung weder in Status noch im Graph angezeigt.

2021-12-28 12:37:41,796 - {/var/www/html/openWB/packages/helpermodules/log.py:38} - ERROR - Payload ungueltig: Topic openWB/set/pv/2/get/power, Payload -857.0 sollte ein In
t sein.
2021-12-28 12:37:41,810 - {/var/www/html/openWB/packages/helpermodules/log.py:38} - ERROR - Unbekanntes set-Topic: openWB/set/pv/2/get/currents, [0, 0, 0]
2021-12-28 12:37:46,485 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - MQTT-Ladepunkte müssen nicht ausgelesen werden.
2021-12-28 12:37:46,486 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - Start device reading {'component1': <modules.json.counter.JsonCounter object at
0x72440bc8>, 'component2': <modules.json.inverter.JsonInverter object at 0x72440d78>}
2021-12-28 12:37:49,487 - {/var/www/html/openWB/packages/helpermodules/log.py:38} - ERROR - Thread-4401 konnte nicht innerhalb des Timeouts die Werte abfragen, die abgefrag
ten Werte werden nicht in der Regelung verwendet.
2021-12-28 12:37:49,491 - {/var/www/html/openWB/packages/helpermodules/log.py:47} - ERROR - Fehler im allgemeinen PV-Modul fuer pv2
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/control/pv.py", line 38, in calc_power_for_all_components
    self.data["get"]["power"] += data.data.pv_data[module].data["get"]["power"]
KeyError: 'power'
2021-12-28 12:37:49,666 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - Antwort auf http://raspi3.fritz.box:8081: { "version": "0.8.0", "generator": "vz
logger", "data": [ { "uuid": "8cf75360-d429-11eb-9433-7b3df3e62eb7", "last": 1640691468952, "interval": -1, "protocol": "sml", "tuples": [ [ 1640691466954, 6500821.90000000
04 ] ] }, { "uuid": "01aae160-d42a-11eb-b856-6363747dc3d8", "last": 1640691468952, "interval": -1, "protocol": "sml", "tuples": [ [ 1640691466954, 7289848.2000000002 ] ] },
 { "uuid": "1d3ac010-d42a-11eb-ae8f-4799bf1e45fd", "last": 1640691468952, "interval": -1, "protocol": "sml", "tuples": [ [ 1640691466954, 18989925.400000002 ] ] }, { "uuid"
: "3193e720-d42a-11eb-8461-cf883a763221", "last": 1640691468952, "interval": -1, "protocol": "sml", "tuples": [ [ 1640691466954, 1679310.5 ] ] }, { "uuid": "0265cf40-d42b-1
1eb-b7e7-3ff66060fa6b", "last": 1640691468952, "interval": -1, "protocol": "sml", "tuples": [ [ 1640691466954, -170 ] ] }, { "uuid": "e0f95960-d42a-11eb-bb4a-05a2a3f49d4f",
 "last": 1640691468952, "interval": -1, "protocol": "sml", "tuples": [ [ 1640691466954, 13790670.200000001 ] ] }, { "uuid": "aa5aac00-d42a-11eb-b608-d568343b9bdc", "last":
1640691468952, "interval": -1, "protocol": "sml", "tuples": [ [ 1640691466954, 20669236 ] ] }, { "uuid": "b2138ca0-d42b-11eb-ad2b-4338d0cd21d5", "last": 1640691468819, "int
erval": -1, "protocol": "sml", "tuples": [ [ 1640691466827, 1510.3000000000002 ] ] }, { "uuid": "c5501860-1b10-11ec-b4f4-f33be250f5db", "last": 1640691468819, "interval": -
1, "protocol": "sml", "tuples": [ [ 1640691466827, 26323077.900000002 ] ] }, { "uuid": "8ae19880-d42b-11eb-8e73-15f0bf1b0d58", "last": 1640691468819, "interval": -1, "proto
col": "sml", "tuples": [ [ 1640691466827, -859 ] ] }, { "uuid": "47cc29e0-8f20-11eb-835c-79ab7da12345", "last": 1640691468221, "interval": 5, "protocol": "exec", "tuples":
[ [ 1640691468221, 360 ] ] } ] }
2021-12-28 12:37:49,667 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - Komponente Json Zähler auslesen.
2021-12-28 12:37:50,430 - {/var/www/html/openWB/packages/helpermodules/log.py:32} - INFO - # ***Start***
2021-12-28 12:37:50,431 - {/var/www/html/openWB/packages/helpermodules/log.py:47} - ERROR - Fehler in der allgemeinen Ladepunkt-Klasse fuer Ladepunkt cp0
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/control/chargepoint.py", line 100, in get_power_counter_all
    counter_all = counter_all + chargepoint.data["get"]["counter"]
KeyError: 'counter'
2021-12-28 12:37:50,799 - {/var/www/html/openWB/packages/helpermodules/log.py:32} - INFO - LP 0: Keine Ladung, da kein Auto angesteckt ist.
2021-12-28 12:37:50,801 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - Kein PV-Modul konfiguriert.
2021-12-28 12:37:50,803 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - all
{'get': {'power': 0}, 'config': {'configured': False}, 'set': {'charging_power_left': 0, 'switch_on_soc_reached': 0, 'hybrid_system_detected': False}}
2021-12-28 12:37:50,805 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - all
{'get': {'daily_yield': 0, 'power_all': 0, 'counter_all': 0}}
2021-12-28 12:37:50,806 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - cp0
{'set': {'charging_ev': -1, 'charging_ev_prev': -1, 'autolock_state': 0, 'current': 0, 'energy_to_charge': 0, 'plug_time': '0', 'rfid': 0, 'manual_lock': False, 'log': {'co
unter_at_plugtime': 0, 'timestamp_start_charging': '0', 'counter_at_mode_switch': 0, 'charged_since_mode_switch': 0, 'charged_since_plugged_counter': 0, 'range_charged': 0,
 'time_charged': '00:00', 'chargemode_log_entry': '_'}}, 'get': {'read_tag': {'tag': '0', 'timestamp': '0'}, 'daily_yield': 0, 'plug_state': False, 'charge_state': False, '
power_all': 0, 'connected_vehicle': {'soc_config': {}, 'soc': {'range': 0, 'range_unit': 'km'}, 'info': {'id': 0, 'name': 'Standard-Fahrzeug'}, 'config': {'charge_template'
: 0, 'ev_template': 0, 'chargemode': 'stop', 'priority': False, 'current_plan': '', 'average_consumption': 17}}, 'state_str': 'Keine Ladung, da kein Auto angesteckt ist.'},
 'config': {'name': 'Standard-Ladepunkt', 'ev': 0, 'template': 0, 'connected_phases': 3, 'phase_1': 0, 'auto_phase_switch_hw': False, 'control_pilot_interruption_hw': False
, 'id': 0, 'connection_module': {'type': 'mqtt', 'configuration': {}}, 'power_module': {}}}
2021-12-28 12:37:50,806 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - cpt0
{'name': 'Standard Ladepunkt-Vorlage', 'autolock': {'wait_for_charging_end': False, 'active': False, 'plans': {}}, 'rfid_enabling': False, 'valid_tags': []}
2021-12-28 12:37:50,806 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - all
{'set': {'loadmanagement_active': False, 'loadmanagement_available': True, 'home_consumption': 4795, 'invalid_home_consumption': 1, 'daily_yield_home_consumption': 0}, 'get
': {'hierarchy': [{'id': 'counter1', 'children': []}, {'id': 'cp0', 'children': []}]}}

LKuemmel, 2021-12-29:
In PR #234 habe den Datentyp für die Leistungen auf float umgestellt. Damit sollte es funktionieren.

DetMoerk, 2021-12-29:
Hallo Lena, Leistung sieht jetzt gut aus, aber der Counter Wert kommt nicht.
`2021-12-29 17:12:20,663 - {/var/www/html/openWB/packages/helpermodules/log.py:47} - ERROR - Fehler im allgemeinen PV-Modul fuer pv1
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/control/pv.py", line 39, in calc_power_for_all_components
    self.data["get"]["counter"] += data.data.pv_data[module].data["get"]["counter"]
KeyError: 'counter'
2021-12-29 17:12:20,678 - {/var/www/html/openWB/packages/helpermodules/log.py:38} - ERROR - Payload ungueltig: Topic openWB/set/pv/1/get/counter, Payload 26328845.4 sollte ein Int sein.
2021-12-29 17:12:20,680 - {/var/www/html/openWB/packages/helpermodules/log.py:38} - ERROR - Unbekanntes set-Topic: openWB/set/pv/1/get/currents, [0, 0, 0]
`

LKuemmel, 2021-12-30:
So, ich habe jetzt nochmal alle Topics der Module geprüft. Jetzt sollte es aber passen.

DetMoerk, 2021-12-30:
Sieht gut aus. Habe das round entfernt und ich bekomme jetzt auch im PV Counter eine Nachkommastelle in den Wh.

Der Error mit den Currents ist auch weg. 

Sehr Cool. Danke

closed 2021-12-30 by DetMoerk