openWB/core #230: Geräte mit mehreren Zählertypen (Fronius)

vuffiraa72, 2021-12-28:
Bisher habe ich gesehen, dass es nur beim Fronius WR unterschiedliche Zählertypen gibt: `counter_s0` und `counter_sm`.
Bei anderen Geräten wird der Zähler immer als Typ `counter` definiert.

In der weiteren Logik wird damit unterschiedlich umgegangen, einmal wird geprüft, ob im Typ das Wort `counter` enthalten ist, aber es wird auch nur auf `counter` geprüft. Dadurch kann man zwar einen Fronius -Zähler konfigurieren, aber er taucht dann nicht auf der Einstiegsseite auf und wird auch nicht beim Status ausgeführt. In der Konfiguration ist er aber vorhanden und es werden Daten abgefragt

Weitere Sonderfälle sind eventuell `cp_counter` und `evu_counter`.

Die Frage ist also, soll es unterschiedliche Typenbezeichnungen geben, oder müssen Geräte mit mehreren Zählern, WRs und Speichern umgehen können, wenn die sich nicht direkt am Gerätetyp unterscheiden lassen?

hhoefling, 2022-01-05:
Ich  bin kein python Spezialist. Aber sollte man nicht prüfen ob was von der basisklasse Counter abgeleitet ist statt eines textvergleiches

LKuemmel, 2022-01-14:
Das Issue ist schon eine Weile offen... wir waren nicht untätig, sondern haben verschiedene Lösungsansätze durchdacht. Allerdings war keiner der Lösungsansätze trivial, sodass es erstmal einen Workaround für das Fronius-Modul gibt.

LKuemmel, 2022-01-14:
@hhoefling Ja, innerhalb des Devices wird mit `isinstance` geprüft, ob von der entsprechenden Counterklasse abgeleitet wurde.
Dies ist jedoch nicht an allen Stellen möglich, da zB das Frontend gar nicht mit Instanzen arbeitet.

closed 2022-01-27 by benderl