openWB/core #231: Inkonsistente MQTT Topic-Namen

vuffiraa72, 2021-12-28:
Die Topic-Namen sollten vereinheitlicht werden, also entweder Singular oder Plural. Die Statusseite liefert momentan nicht alle bekannten Werte. In den Statusklassen werden vorzugsweise Pluralbegriffe verwendet, die MQTT-Topics folgen eher Singular-Begriffen. Besonders `current` und `power_factor` werden inkonsistent verwendet.

_chargepoint.py
```python
pub_to_broker("openWB/set/chargepoint/" + str(self.num) + "/get/voltage", state.voltages, 2)
pub_to_broker("openWB/set/chargepoint/" + str(self.num) + "/get/current", state.currents, 2)
pub_to_broker("openWB/set/chargepoint/" + str(self.num) + "/get/power_factor", state.power_factors, 2)
```

_counter.py
```python
pub_to_broker("openWB/set/counter/" + str(self.num) + "/get/voltage", counter_state.voltages, 2)
pub_to_broker("openWB/set/counter/" + str(self.num) + "/get/current", counter_state.currents, 2)
pub_to_broker("openWB/set/counter/" + str(self.num) + "/get/power_phase", counter_state.powers, 2)
pub_to_broker("openWB/set/counter/" + str(self.num) + "/get/power_factors", counter_state.power_factors, 2)
```

_inverter.py
```python
pub_to_broker("openWB/set/pv/" + str(self.num) + "/get/currents", inverter_state.currents, 1)
```

Status.vue
```vue
'openWB/counter/' + counter.id + '/get/voltage'

'openWB/counter/' + counter.id + '/get/current'

'openWB/counter/' + counter.id + '/get/power_phase'

'openWB/counter/' + counter.id + '/get/power_factor'
```

LKuemmel, 2022-01-04:
#240 bitte mal reinschauen. Entsprechende PRs für UI und Module im snaptec-Repo folgen noch.

DetMoerk, 2022-01-05:
Ich finde das passend, was in #240 eingebaut ist.

vuffiraa72, 2022-01-05:
Mir gefällt #240 auch. Die Begrifflichkeit ordentlich sich an der Payload.

DetMoerk, 2022-01-07:
#240 it Gemenges, dann kann das hier wohl zu gemacht werden oder?
VG
Det

closed 2022-01-11 by LKuemmel