openWB/core #236: Logging

DetMoerk, 2021-12-30:
Hallo openWB Team, 
Ich finde das Logging in der main.log etwas unübersichtlich. Die Info, aus welcher Zeile im log.py der Fehler geworfen wird ist meiner Meinung nach nicht sehr hilfreich und macht die Zeile nur breit. Dagegen bleibt der Hinweis, an welcher Stelle im Code der Fehler geworfen wird, verborgen. Ich vermute, bei einer direkten Verwendung der Python Loggers Klasse wäre das nicht so.

VG
Det

LKuemmel, 2022-01-04:
Gefixt in #243: Das war ursprünglich auch so umgesetzt. Ist aber im Zuge der Überarbeitung untergegangen. Ich habe auch das /var/www/html/openWB aus der Pfadangabe entfernt. Dateien aus anderen Ordnern werden aktuell nicht aufgerufen und so sind die Zeilen nicht so breit.

DetMoerk, 2022-01-04:
Sehr cool. Damit kann man wesentlich besser verfolgen was passiert.

closed 2022-01-04 by DetMoerk