openWB/core #239: Konfigurationsabschluss PV-Komponente

vuffiraa72, 2022-01-02:
Nach Abschluss der Konfiguration einer PV-Komponente erwarte ich, dass das im MQTT-Topic "openWB/pv/config/configured" hinterlegt wird. Meine Installation weigert sich aber.

Soweit ich das im Code überblicke, sollte das Umsetzen des Konfigurationsstatus in der Datei "packages/main.py" erledigt werden. Die interessante Funktion ist wohl "handler_with_control_interval".

In Zeile 55 werden die Konfigurationsdaten überprüft:
```python
loadvars.get_virtual_values()
```

Letztendlich wird die Funktion "calc_power_for_all_components" aus der Datei "packages/control/pv.py" gerufen. Hier wird in Zeile 48 auch der Konfigurationsstaus gesetzt:
```python
self.data["config"]["configured"] = True
```

Danach werden in der Datei "packages/main.py" in Zeile 58 die Werte in "self.data" wieder mit den Daten aus dem Broker überschrieben.
```python
prep.copy_module_data()
```

Theoretisch wird der Konfigurationsstatus dann in "packages/main.py" Zeile 70 als Teil der Funktion "prep.setup_algorithm" veröffentlich. Hier wird dann aber nur veröffentlicht, was dem Broker schon bekannt war.
```python
prep.setup_algorithm()
```

Ich sehen folgende Möglichkeiten:
1. Ich habe etwas übersehen.
1. Der Konfigurationsstatus könnte innerhalb der Funktion "calc_power_for_all_components" veröffentlich werden
1. Beim Lesen der Broker-Daten ist an der genannten Stelle der Datei "packages/main.py" das Lesen des Konfigurationsstatus zu übergehen.

Ansonsten hat der Code für die PV-Komponente sehr große Ähnlichkeit mit dem der Speicher-Komponente. Dort sehe ich das gleiche Problem, kann aus aber wegen des fehlenden Speichers in meiner Installation nicht direkt prüfen.

LKuemmel, 2022-01-04:
Danke für den Bugreport und deine Recherche.
Ich konnte den Fehler nachstellen und habe es wie in Punkt 2 vorgeschlagen gelöst.

hhoefling, 2022-01-05:
Das ist  dann aber eine Nebenwirkungen die ich bei diesem methodennahmen nicht erwarten würde

vuffiraa72, 2022-01-05:
Die Lösung funktioniert.

Vielleicht kann man hier noch vermerken, dass dadurch nicht nur die Konfiguration der PV-Komponente bestätigt wird, sondern die Ermittlung der Werte selber. Diese Funktionalität bestätigt dadurch regelmäßig, dass gültige Werte vorhanden sind. Für mich passt es zum Methodennamen.

closed 2022-01-06 by LKuemmel