openWB/core #241: EVU Kit Flex mit SDM630 fehlende Daten

DetMoerk, 2022-01-04:
Ich habe seit heute einen PE11 und einen SDM72 (analog SDM630) für meine WP in Betrieb. Um zu testen habe ich den SDM auch in der 2.0 Alpha als 2. EVU Modul eingestellt. Analog auch auf meinen 1.9 Entwicklungssystem als EVU flex Kit. 

Bei der 1.9 kommen alle Daten für alle Phasen an, also Spannung, Stromstärke, Leistung und Power Faktor. Die Summierten Werte Gesamtleistung, Bezug, Einspeisung (ist natürlich 0, da nur bezogen wird), Frequenz und Schieflast sind ebenfalls vorhanden.

In der 2.0 Alpha kommt bei den Phasenwerten nur die Spannung einer Phase (0,0,230.14) Strom (0,0,0) ist falsch, Leistung (0,0,0) ist ebenfalls falsch, Leistungsfaktor ist komplett leer. Import, Leistug Netzfrequenz passen. 

VG
Det

LKuemmel, 2022-01-04:
Kannst Du bitte mal einen Auszug aus dem Log einstellen? Kommen da irgendwelche Fehlermeldungen?

DetMoerk, 2022-01-04:
Sorry, kann geschlossen werden, ich hatte den Service vor dem git pull beendet, die Daten die ich abgelesen habe waren noch alt.

Passt alles, außer dem Power Faktor, der nicht angezeigt wird.

LKuemmel, 2022-01-04:
Alles klar.

Stimmt, da ist noch das Topic falsch.

DetMoerk, 2022-01-04:
Auch wenn hier zu ist, ich hab gerade im Log gesehen, dass da noch ein PV2 angefragt wird obwohl ich das PV2 Modul nur zum Test angelegt hatte. 
Kann man das irgendwie bereinigen?

VG
Det

DetMoerk, 2022-01-04:
Bekomme die folgenden Meldungen regelmäßig im Log. PV3 habe ich gerade angelegt, aber auch wieder gelöscht. 

Irgendwas bleibt da noch bestehen.

`2022-01-04 15:32:20,373 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - pv2
{'get': {'daily_yield': 0, 'monthly_yield': 0, 'yearly_yield': 0, 'fault_str': 'TCP-Client konnte keine Verbindung zu 192.168.193.15:8899 aufbauen. Bitte Einstellungen (IP-Adresse, ..) und Hardware-Anschluss pruefen.', 'fault_state': 2}}
2022-01-04 15:32:20,373 - {/var/www/html/openWB/packages/helpermodules/log.py:35} - DEBUG - pv3
{'get': {'daily_yield': 0, 'monthly_yield': 0, 'yearly_yield': 0}}
`

VG
Det

DetMoerk, 2022-01-04:
Nach einem Restart des openWB Services bekomme ich nur noch die Meldung von der PV2. Kann es sein, dass es daher kommt, dass hier noch eine Fehlermeldung (FaultState) im MQTT Topic hängt. 
Die PV2 konnte, so wie konfiguriert, in meiner Umgebung nicht funktionieren. Daher kam der Fault State. Evtl. Wurde der nicht gelöscht, als die Komponente gelöscht wurde.

VG
Det

LKuemmel, 2022-01-04:
Die Komponente wird offenbar nicht komplett gelöscht. Ich sehe es mir gerade an...

DetMoerk, 2022-01-04:
Ich habe jetzt mal in den Topics geschaut, 

Es sind von PV2 noch 

openWB/pv/2/get/fault_state
Und

openWB/pv/2/get/fault_str

In den Topics.

VG
Det

LKuemmel, 2022-01-04:
Das Modul wurde nicht vollständig im Algorithmus gelöscht. Mit PR #242 funktioniert es bei mir jetzt.

DetMoerk, 2022-01-04:
Hmm, ich sehe keine Aufrufe mehr in der main.log, aber ich habe jetzt trotzdem noch Topics im MQTT für PV/3 und PV/4. 

Und für den PV/2 bekomme ich den Fehler weiterhin. 

VG
Det

LKuemmel, 2022-01-04:
Hast du außer git pull noch etwas gemacht? Die Regelung erkennt nicht, dass da korrupte Konfigurationen auf dem Broker liegen. Die müsstest du einmal manuell bereinigen. 
Also den Service stoppen, alle Topics unter `openWB/pv/2, openWB/pv/3, openWB/pv/4` löschen und dann den Service wieder starten.

DetMoerk, 2022-01-04:
Ja, ich stoppe den Service, mache dann den git pull und starte den Service dann wieder. Ich lösche jetzt mal den PV2 Zweig raus. Die anderen sind ja raus, wurden aber in dem Tool auf dem iPAD noch angezeigt.

VG
Det

DetMoerk, 2022-01-04:
Strange, jetzt habe ich die PV/2 Topics aus dem MQTT gelöscht und neu gestartet, aber der Fehler mit pv2 kommt weiterhin im main.log. Wird das im _local MQTT auch noch irgendwo gespeichert?

VG
Det

LKuemmel, 2022-01-04:
Du musst es im Broker localhost:1886 löschen. Die Brücke funktioniert nur in diese Richtung.
Sorry, hätte ich dazu schreiben sollen.

Ist unter `system/device/x/component/x` noch eine Komponente mit ID 2 drin? Dann diese auch noch löschen. 
Das manuelle Löschen funktioniert übrigens nur mit WR und Bat. Zähler müssen auch noch aus der Hierarchie entfernt werden.

DetMoerk, 2022-01-04:
OK, habe mir gerade den Local Mosquitto auch so eingerichtet, dass ich ihn von außen sehen kann. Da war die PV2 noch drin. Jetzt ist mein System wieder sauber :-) und sollte es auch bleiben.

VG und schönen Feierabend
Det

closed 2022-01-04 by DetMoerk