openWB/core #245: Speicher Daten bleiben nach Löschen noch aktiv

DetMoerk, 2022-01-05:
@LKuemmel: Hi Lena, ich habe gestern noch ein JSON Bat Modul eingerichtet, und auch wieder gelöscht. Es bleibt unter openWB/bat/get noch der Power und der SoC bestehen. Auf dem Frontend sind die Daten daher auch noch sichtbar.
Kannst du dir das bitte anschauen?
Im Status ist der Speicher nicht angezeigt.

VG
Det

LKuemmel, 2022-01-07:
Hallo Det,
das configured-Topic wurde nicht zurückgesetzt, wenn alle Speicher/WR wieder gelöscht wurden. Ist in PR #253 gefixt.

DetMoerk, 2022-01-07:
Gelöscht wird jetzt, das habe ich gerade getestet, aber ich konnte unter Flex Kit keinen Speicher anlegen, der Daten gelesen hätte. Welche Version sollte denn einen SDM630 auslesen können? Ich habe 1-3 versucht, aber bei keinem habe ich eine Leistung erhalten. 

VG
Det

LKuemmel, 2022-01-11:
Version 2 liest den SDM630 aus. Kommen Fehlermeldungen im Log?

DetMoerk, 2022-01-11:
Die Fehlermeldungen passen irgendwie nicht. Bei mir springt er im openwb_flex/bat.py aus nach dem finally (Zeile 48) und anschließendem close_connection raus und geht nicht mehr zu bat_state= ...

``` try:
            imported = self.__client.get_imported()
            exported = self.__client.get_exported()
            if self.component_config["configuration"]["version"] == 2:
                _, power = self.__client.get_power() * -1
            else:
                _, power = self.__client.get_power()
        finally:
            self.__tcp_client.close_connection()
        bat_state = BatState(
            imported=imported,
            exported=exported,
            power=power```

Vermute mein Entwicklungssystem ist kaputt getestet.

DetMoerk, 2022-01-11:
```
2022-01-11 13:11:41,309 - {/packages/modules/common/fault_state.py:35} - ERROR - Speicher-Kit flex SDM630: FaultState FaultStateLevel.ERROR, FaultStr <class 'ValueError'> not enough values to unpack (expected 2, got 0), Traceback:
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/openwb_flex/device.py", line 59, in update
    self._components[component].update()
  File "/var/www/html/openWB/packages/modules/openwb_flex/bat.py", line 44, in update
    _, power = self.__client.get_power() * -1
ValueError: not enough values to unpack (expected 2, got 0)

2022-01-11 13:11:41,314 - {/packages/control/bat.py:74} - ERROR - Fehler im Bat-Modul bat3
```

LKuemmel, 2022-01-11:
Besteht die Fehlermeldung noch, nachdem Du den Broker zurückgesetzt hast?

DetMoerk, 2022-01-11:
Ja, bin gerade am schauen. Mir kommt es hier so vor, als ob es auch mit er Umbenennung von power/powers was zu tun hat. Wenn ich im VSCode Debugge, fliegt er mir in control/bat.py in Zeile 66 mit Value Error raus.

LKuemmel, 2022-01-11:
Ja, die Leistung pro Phase wird bei der Batterie nicht benötigt. Führt der Versuch, die Register für die Leistung pro Phase auszulesen vielleicht zu einem Fehler und es gibt deshalb auch keine Leistung? (Ich kann das mangels Hardware nicht testen.)

DetMoerk, 2022-01-11:
Nee, es ist was anderes, das * -1 funktioniert hier wohl nicht. Das geht an dem powers Array in die Hose.

LKuemmel, 2022-01-11:
Ich korrigiere das eben.

DetMoerk, 2022-01-11:
```
 if self.component_config["configuration"]["version"] == 2:
                #_, power = self.__client.get_power() * -1
                _, power = self.__client.get_power() 
                power = power * -1
```

DetMoerk, 2022-01-11:
oder besser so

```
        try:
            imported = self.__client.get_imported()
            exported = self.__client.get_exported()
            _, power = self.__client.get_power() 
            if self.component_config["configuration"]["version"] == 2:
                power = power * -1
        finally:
```

LKuemmel, 2022-01-11:
Jetzt ist es schon drin...

DetMoerk, 2022-01-11:
OK, ich pulle mal auf mein System

DetMoerk, 2022-01-11:
Sieht gut aus.

fibroin, 2025-04-09:
Guten Tag,
ich habe zum Testen einen mqtt-Speicher eingerichet. Nach dem Entfernen bleibt dieser auf dem Frontend angezeigt.
Das bat/config/configured-Topic wird nicht auf False gesetzt.
Können Sie das bitte nochmal prüfen?
Gruß
Andreas

LKuemmel, 2025-04-11:
Bitte bei Problemen immer einen ungekürzten Logauszug posten:
Dazu unter System->Fehlersuche das Debuglevel auf Details stellen und mindestens zwei komplette Durchläufe von # ***Start*** bis # ***Start*** aus dem Main-Log kopieren, während das Problem auftritt. Sensible Daten wie Benutzernamen und Kennwörter unkenntlich machen.
Alternativ das Log hier: https://paste.openwb.de/ hochladen und den Link im Thread posten. Die Logs werden dort automatisch nach 90 Tagen gelöscht. Bitte keine gezippten Logfiles hochladen!

fibroin, 2025-04-14:
Hallo @LKuemmel,
der Logauszug ist hier gespeichert: [https://paste.openwb.de/ZknaT8mbhruYVON](https://paste.openwb.de/ZknaT8mbhruYVON)
Ich habe eine relativ einfache Config mit 2 Wechselrichtern und einem MQTT-Zähler.
Als Test hatte ich einen MQTT-Speicher eingebunden. Den habe ich per mqtt-Explorer auf 50% Ladestand gesetzt. Danach habe ich diesen wieder entfernt.
Auf dem Frontend bleibt der Speicher aber erhalten.
Gruß
Andreas