openWB/core #247: Anlegen / Löschen von Ladepunkten

DetMoerk, 2022-01-05:
Leider bleiben auch von den Ladepunkten Einträge im Mosquitto, wenn sie wieder gelöscht werden. 
Bei Add eines Ladepunktes ip_evse wird der folgende Fehler geworfen.

`Bei der Verarbeitung des Befehls 'addChargepoint' mit den Parametern '{ "type": "ip_evse" } ' ist ein Fehler aufgetreten:

Es ist ein interner Fehler aufgetreten: Traceback (most recent call last):
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 108, in on_message
    func(connection_id, payload)
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 164, in addChargepoint
    module = importlib.import_module("."+payload["data"]["type"]+".chargepoint_module", "modules")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/var/www/html/openWB/packages/modules/ip_evse/chargepoint_module.py", line 10, in <module>
    from packages.modules.common.component_state import ChargepointState
ModuleNotFoundError: No module named 'packages'
`

LKuemmel, 2022-01-06:
Bei Anlegen des IP-EVSE war noch ein Fehler im Import. Ist in PR #250 korrigiert.

Das LP-Topics bestehen bleiben, konnte ich nicht beobachten. Welche LP hast du angelegt und gelöscht? Was ist noch übrig geblieben auf dem Broker?

DetMoerk, 2022-01-07:
Hallo Lena, 
der IP EVSE hat jetzt funktioniert, lässt sich auch wieder rückstandsfrei löschen. 
Einen MQTT habe ich angelegt und gelöscht, da sind dann noch die 
openWB/chargepoint/2/set/charging_ev
openWB/chargepoint/2/set/current
openWB/chargepoint/2/set/energy_to_charge 
Topics übrig nach dem Löschen.

Bei der openWB Pro sieht das für mich auch so aus, dass die identischen Topics bleiben. 
Ich bin auf dem _local Mosqiutto. Habe jetzt auch auf dem anderen geschaut, sieht identisch aus.

aber irgendwie habe ich das Gefühl, dass da noch was größeres nicht läuft. Ich habe jetzt auch bei einem Externe openWB einen LP angelegt und gelöscht, das ging meiner Meinung nach vorher, jetzt habe ich hier noch Fragmente auch im Get Zweig (fault_str und fault_state). 

Das Verhalten ist da sowieso etwas undurchsichtig. Es gibt einen Speichern Button unten an der LP Konfiguration. Wenn man aber einen LP anlegt, werden sofort die MQTT Topics angelegt, auch bereits vor dem Speichern. Rückgängig geht dann nicht mehr. Vermute der Speichern Button wird nur benötigt, wenn man dann Änderungen im Formular macht, wie z.B. LP umbenennen. 
VG
Det

DetMoerk, 2022-01-11:
@benderl : Ichhabe jetzt im Frontend noch einige Zeilen für Ladepunkte, die ich nicht zuordnen kann. Alle sehen identisch aus mit -- als Bezeichner. Hast du einen Tipp, wie ich die wieder weg bekomme?

LKuemmel, 2022-01-11:
Durch die Bugs bei den Ladepunkten sind die LP nicht richtig aus der Hierarchie gelöscht worden. 
Wahrscheinlich werden die auch noch unter Konfiguration Lastmanagement -> Struktur angezeigt? Wenn du auf das i bei der Struktur klickst, wird unten der Payload angezeigt. Den korrigieren und an `openWB/set/counter/get/hierarchy` publishen.

DetMoerk, 2022-01-11:
Ich habe den Eindruck, dass da bei mir da im Broker so manches durcheinander ist. Bin gerade am überlegen, ob ich mein Entwicklungssystem platt mache und noch mal von vorne starte, da ich keine Idee habe, wie ich die Broker sauber bekommen soll. 
Oder gibt es da einen Tipp @LKuemmel @benderl ?

benderl, 2022-01-11:
Den openwb2 Service beenden und dann das Skript "reset-broker.sh" mit sudo ausführen. Dadurch startest Du wieder mit einer leeren Konfiguration.

DetMoerk, 2022-01-11:
> Den openwb2 Service beenden und dann das Skript "reset-broker.sh" mit sudo ausführen. Dadurch startest Du wieder mit einer leeren Konfiguration.

Hat funktioniert. Danke für den Tipp