openWB/core #265: Fehler im Lastmanagement Modul

DetMoerk, 2022-01-11:
Ich bekomme einen Fehler im Lastmanagement Modul. Referenziert wird der Zähler 4 (mein VZLogger per JSON). Kann es sein, dass es hier auf Grund von fehlenden Einzelphasen eine Meldung gibt?

Ich versuch mal noch mehr raus zu finden.

VG
Det

DetMoerk, 2022-01-11:
in control/counter.py Zeile 367 sollte wohl nicht fix auf counter/0/ gesetzt werden oder? Oder ist counter/0 ein spezieller counter? 

```
    def put_stats(self):
        try:
            Pub().pub("openWB/set/counter/0/set/consumption_left", self.data["set"]["consumption_left"])

```

LKuemmel, 2022-01-11:
Am Anfang war der Zähler 0 der EVU-Zähler. Mittlerweile wird der EVU-Zähler aus der Hierarchie ermittelt. Das war wohl noch ein Relikt.

DetMoerk, 2022-01-11:
Yeah, jetzt kommen bei mir keine ERROR Meldungen mehr :-).

closed 2022-01-11 by DetMoerk