openWB/core #267: Fehler im Werte-Loggingmodul

DetMoerk, 2022-01-11:
```
2022-01-11 16:39:56,296 - {/packages/helpermodules/measurement_log.py:268} - ERROR - Fehler im Werte-Loggingmodul
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/helpermodules/measurement_log.py", line 232, in update_daily_yields
    daily_yield = data.data.pv_data[pv].data["get"]["counter"] - \
KeyError: 'pv1'

```

DetMoerk, 2022-01-11:
Hänge ich morgen an.

DetMoerk, 2022-01-12:
```
2022-01-11 18:05:55,779 - {/packages/main.py:39} - INFO - # ***Start***
2022-01-11 18:05:55,785 - {/packages/modules/json/device.py:55} - DEBUG - Start device reading {'component4': <modules.json.counter.JsonCounter object at 0x739e3940>, 'component5': <modules.json.inverter.JsonInverter object at 0x739e3af0>}
2022-01-11 18:05:55,812 - {/packages/modules/common/req.py:10} - DEBUG - Get-Response: { "version": "0.8.0", "generator": "vzlogger", "data": [ { "uuid": "8cf75360-d429-11eb-9433-7b3df3e62eb7", "last": 1641920755467, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755467, 6744538.9000000004 ] ] }, { "uuid":
"01aae160-d42a-11eb-b856-6363747dc3d8", "last": 1641920755467, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755467, 7631771.9000000004 ] ] }, { "uuid": "1d3ac010-d42a-11eb-ae8f-4799bf1e45fd", "last": 1641920755467, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755467, 18998919.900000002 ] ] }, {
 "uuid": "3193e720-d42a-11eb-8461-cf883a763221", "last": 1641920755467, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755467, 1686783.7000000002 ] ] }, { "uuid": "0265cf40-d42b-11eb-b7e7-3ff66060fa6b", "last": 1641920755467, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755467, 794 ] ] }, { "uuid
": "e0f95960-d42a-11eb-bb4a-05a2a3f49d4f", "last": 1641920755467, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755467, 14376310.800000001 ] ] }, { "uuid": "aa5aac00-d42a-11eb-b608-d568343b9bdc", "last": 1641920755467, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755467, 20685703.700000003 ] ] }
, { "uuid": "b2138ca0-d42b-11eb-ad2b-4338d0cd21d5", "last": 1641920755375, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755375, 1513.2 ] ] }, { "uuid": "c5501860-1b10-11ec-b4f4-f33be250f5db", "last": 1641920755375, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755375, 26401706.200000003 ] ] }, {
 "uuid": "8ae19880-d42b-11eb-8e73-15f0bf1b0d58", "last": 1641920755375, "interval": -1, "protocol": "sml", "tuples": [ [ 1641920755375, 0 ] ] } ] }
2022-01-11 18:05:55,814 - {/packages/modules/json/counter.py:35} - DEBUG - Komponente Json Zähler auslesen.
2022-01-11 18:05:56,883 - {/packages/modules/json/inverter.py:34} - DEBUG - Komponente Json Wechselrichter auslesen.
2022-01-11 18:05:58,114 - {/packages/control/counter.py:358} - DEBUG - 10206W EVU-Leistung, die noch bezogen werden kann.
2022-01-11 18:05:58,118 - {/packages/control/pv.py:85} - DEBUG - -909.0W EVU-Überschuss, der für die Regelung verfügbar ist, davon 0W für die Einschaltverzögerung reservierte Leistung.
2022-01-11 18:05:58,133 - {/packages/control/data.py:285} - DEBUG - all
{'get': {'power': 0, 'imported': 0, 'exported': 0, 'daily_yield_export': 0, 'daily_yield_import': 0, 'soc': 0}, 'config': {'configured': False}, 'set': {'charging_power_left': 0, 'switch_on_soc_reached': 0, 'hybrid_system_detected': False}}
2022-01-11 18:05:58,139 - {/packages/control/data.py:285} - DEBUG - all
{'get': {'daily_yield': 0, 'power': 0, 'counter': 0}}
2022-01-11 18:05:58,139 - {/packages/control/data.py:285} - DEBUG - cpt0
{'name': 'Standard Ladepunkt-Vorlage', 'autolock': {'wait_for_charging_end': False, 'active': False, 'plans': {}}, 'rfid_enabling': False, 'valid_tags': []}
2022-01-11 18:05:58,140 - {/packages/control/data.py:285} - DEBUG - cpt1
{'name': 'Standard Ladepunkt-Vorlage', 'autolock': {'wait_for_charging_end': False, 'active': False, 'plans': {}}, 'rfid_enabling': False, 'valid_tags': [], 'id': 1}
2022-01-11 18:05:58,140 - {/packages/control/data.py:285} - DEBUG - all
{'set': {'loadmanagement_active': False, 'loadmanagement_available': True, 'home_consumption': 794, 'invalid_home_consumption': 0, 'daily_yield_home_consumption': 0}, 'get': {'hierarchy': [{'id': 'counter4', 'children': []}]}}
2022-01-11 18:05:58,141 - {/packages/control/data.py:285} - DEBUG - counter4
{'set': {'consumption_left': 10206, 'currents_used': [0, 0, 0]}, 'get': {'daily_yield_export': 0, 'daily_yield_import': 0, 'fault_str': 'Kein Fehler.', 'fault_state': 0, 'voltages': [0, 0, 0], 'currents': [0, 0, 0], 'powers': [0, 0, 0], 'power_factors': [0, 0, 0], 'imported': 14376310.8, 'exported': 20685703.7000000
03, 'power': 794, 'frequency': 50}, 'config': {'max_currents': [16, 16, 16], 'max_total_power': 11000}}
2022-01-11 18:05:58,143 - {/packages/control/data.py:285} - DEBUG - ct0
{'name': 'Standard-Ladeprofil-Vorlage', 'disable_after_unplug': False, 'prio': False, 'load_default': False, 'time_charging': {'active': False, 'plans': {}}, 'chargemode': {'selected': 'stop', 'pv_charging': {'min_soc_current': 10, 'min_current': 6, 'feed_in_limit': False, 'min_soc': 0, 'max_soc': 100}, 'scheduled_c
harging': {'plans': {}}, 'instant_charging': {'current': 10, 'limit': {'selected': 'none', 'soc': 50, 'amount': 10}}}}
2022-01-11 18:05:58,144 - {/packages/control/data.py:285} - DEBUG - ev0
{'set': {}, 'get': {'range_charged': 0, 'soc': 0}, 'control_parameter': {'required_current': 0, 'phases': 0, 'prio': False, 'timestamp_switch_on_off': '0', 'timestamp_auto_phase_switch': '0', 'timestamp_perform_phase_switch': '0', 'submode': 'stop', 'chargemode': 'stop'}, 'name': 'Standard-Fahrzeug', 'charge_templat
e': 0, 'ev_template': 0, 'tag_id': []}
2022-01-11 18:05:58,144 - {/packages/control/data.py:285} - DEBUG - et0
[... cut ...]
2022-01-11 18:05:58,155 - {/packages/control/algorithm.py:939} - INFO - ## Übrigen Überschuss verteilen.
2022-01-11 18:05:58,160 - {/packages/control/process.py:20} - DEBUG - # Ladung starten.
2022-01-11 18:05:58,164 - {/packages/control/pv.py:429} - DEBUG - -909.0W EVU-Überschuss, der für die Regelung verfügbar ist, davon 0W für die Einschaltverzögerung reservierte Leistung.
2022-01-11 18:05:58,167 - {/packages/control/counter.py:370} - DEBUG - 10206W verbleibende EVU-Bezugs-Leistung

```
Passt das so für die Analyse?
VG
Det

DetMoerk, 2022-01-12:
Sieht so aus, als ob das auch gefixt ist. Ich habe aktuell keine ERRORs mehr im main.log

LKuemmel, 2022-01-12:
Der Fehler ist vermutlich aufgrund des neuen Tages verschwunden. Steht auf der ToDo-Liste...

LKuemmel, 2022-02-03:
Fixed by [d031aec](https://github.com/openWB/core/pull/289/commits/d031aec7cf169a8d7f95bc3976fea70e6670c33d)

closed 2022-02-03 by LKuemmel