openWB/core #277: Range Electric für SOC Ermittlung vorsehen

DetMoerk, 2022-01-16:
In der 1.9 wurde es bereits mehrfach angesprochen, dass die SoC Ermittlung oft auch einen Range in Kilometern übergeben kann. 
Bitte in der V2.0 vorsehen, dass dieser Wert ebenfalls aufgenommen werden kann und im Frontend dargestellt werden kann.

foscoj, 2022-03-19:
Dem Wunsch schließe ich mich an, Likes gibt es hier ja nicht :)

hhoefling, 2022-03-25:
Ich würde auch den akuellen KM-Stand zum PKW Status zählen.
Könnte man im Ladelog gut gerauchen. Genaus wie Start und End Soc der Ladung

DetMoerk, 2022-05-02:
Bei der Mercedes EQ Api kann man über die entsprechenden APIs auch noch mehr Daten raus holen.
651507326000}},{"decklidstatus":{"value":"false","timestamp":1651489687000}},{"doorstatusfrontright":{"value":"false","timestamp":1651507317000}},{"doorstatusfrontleft":{"value":"false","timestamp":1651493107000}},{"doorstatusrearright":{"value":"false","timestamp":1650561850000}},{"windowstatusfrontright":{"value":"2","timestamp":1651152638000}},{"windowstatusfrontleft":{"value":"2","timestamp":1651152638000}},{"windowstatusrearright":{"value":"0","timestamp":1651260119000}},{"windowstatusrearleft":{"value":"0","timestamp":1651260119000}},{"doorstatusrearleft":{"value":"false","timestamp":1650562125000}},{"sunroofstatus":{"value":"1","timestamp":1651507362000}}]
2022-05-02 18:03:29: Lp1: PID:24603: EVP Request: 200
2022-05-02 18:03:29: Lp1: PID:24603: EVP Response: [{"odo":{"value":"22515","timestamp":1651489573000}}]

Der unterste Aufruf wären die Gesamtkilometer. Aber auch Fenster/Tür/Schiebedachstatus sind ermittelbar.

Aber ich glaube das wäre dann etwas Over Engineered

benderl, 2022-05-02:
Dann könnte man, wenn kein Display an der Wallbox ist, durch die Kombination verschiedener offener Fenster und Türen den Lademodus umschalten. Klingt nach DEM Usecase. ;-)

DetMoerk, 2022-05-03:
> Dann könnte man, wenn kein Display an der Wallbox ist, durch die Kombination verschiedener offener Fenster und Türen den Lademodus umschalten. Klingt nach DEM Usecase. ;-)

Ja, das wäre Super. Je nach Öffnung des Fensters wird der Ladestrom bei Min & PV gesteuert. Oder bei Sofort laden.
Aber eigentlich warte ich noch auf die API Funktion, bei der ich die Vorklimatisierung abfragen kann. Damit könnte man dann zur entsprechenden Zeit, die WB freischalten. 
Mal sehen, ob das noch ausgebaut wird. 
https://data.mercedes-benz.com/products/electric-vehicle-status
Da stehen zumindest interessante Dinge drin.
VG
Det

MartinRinas, 2022-09-29:
das müsste mit der aktuellen Implementation des Tesla SoC adressiert sein, ich sehe bei mir neben dem SoC auch die Reichweite:

{"fault_str": "Kein Fehler.", "fault_state": 0, "range_charged": 0, "range_unit": "km", "range": 377.2775762, "soc": 77.0, "timestamp": "09/29/2022, 04:14:37"}

Zu finden in diesem Topic: openWB/chargepoint/<id>/get/connected_vehicle/soc

foscoj, 2022-09-29:
@andig das klingt nach einem guten Andockpunkt für EVCC :)

andig, 2022-09-29:
Absolut. Gibts schon irgendeinen Aufssetzpunkt, ob/wo/wie die Module das liefern können?

foscoj, 2022-09-29:
anscheinend laut dem Tesla Modul: https://github.com/openWB/core/blob/master/packages/modules/tesla/soc.py

self.value_store.set(CarState(soc, range))

andig, 2022-09-29:
Evcc liefert über Kommandozeile. Ich kann gerne ein JSON zurück geben, das bräuchte aber eine Doku wie wo was.

MartinRinas, 2022-09-29:
jup, das ist die 2.0. Bisher gibt es nur das Tesla Modul, die anderen werden wohl nach und nach migriert. Mit dem Tesla-Modul sollen erstmal die Vorraussetzungen, Datenmodell und was es sonst noch so braucht ans fliegen gebracht werden. Was m.E. soweit schon mal ganz gut aussieht.

foscoj, 2022-09-29:
Top, da passt das Multimodul EVCC ja perfekt zu ;)

benderl, 2022-09-30:
> Top, da passt das Multimodul EVCC ja perfekt zu ;)

Nicht wirklich. Die 2.0 ist nicht mehr auf den Einsatz auf einem Raspi begrenzt. Das bedeutet für das kompilierte EVCC Binary, dass es für verschiedene Architekturen vorhanden sein muss.

foscoj, 2022-09-30:
Ich meinte eher die funktionale Sicht.

andig, 2022-09-30:
> > Top, da passt das Multimodul EVCC ja perfekt zu ;)
> 
> Nicht wirklich. Die 2.0 ist nicht mehr auf den Einsatz auf einem Raspi begrenzt. Das bedeutet für das kompilierte EVCC Binary, dass es für verschiedene Architekturen vorhanden sein muss.

Kein Problem, haben wir. Mir fehlt aber weiter die Partnerinformation was zu tun wäre ;)

benderl, 2022-09-30:
Da gibt es noch keine Info, weil die API noch nicht final ist.

benderl, 2022-12-09:
Ich schließe hier mal, da der ursprüngliche Punkt der Reichweite geklärt ist.

closed 2022-12-09 by benderl

andig, 2022-12-09:
@benderl gibts denn jetzt eine Info für die Modulbauer? Im Forum hab ich gesehen, dass einige Module angepasst sind. Gefunden zum "wie" hab ich aber nix.

benderl, 2022-12-09:
Nein, noch nicht. Das hat rleidner auf Eigeninitiative gemacht. Zeigt zumindest, dass die Struktur der Klassen nicht ganz verkehrt umgesetzt wurde und nachvollziehbar ist.