openWB/core #278: MQTT-PV-WR-Werte: Unterschiedliche Topics in Doku und Code

gvzdus, 2022-01-17:
Beim Einrichten eines MQTT-Wechselrichters heisst es in der Beschreibung:

openWB/set/pv/1/power
PV-Leistung in Watt, ohne Nachkommastellen (Integer), nur negativ!
openWB/set/pv/1/counter
Erzeugte Energie in Wh, mit Nachkommastellen (Float), nur positiv

Erwartet werden stattdessen die Topics openWB/set/pv/1/**get**/power und openWB/set/pv/1/**get**/counter

Die Hinweisseite sollte korrigiert werden, allerdings ist im Repository nur generierter Code (web/settings/js/3.js)

benderl, 2022-01-17:
Der Quellcode ist dort zu finden: https://github.com/openWB/openwb-ui-settings

Ich passe die Beschreibung der Topics noch an.

benderl, 2022-01-19:
Ist im Repo openwb-ui-settings angepasst. Kommt beim nächsten Build also hier mit rein.

closed 2022-01-19 by benderl