openWB/core #282: EVU: Ohne currents-Werte keine Steuerung und Exceptions

gvzdus, 2022-01-19:
EVU: Wenn keine currents-Werte kommen, steuert die OpenWB nicht und schreibt Exceptions in der algorith.py ins mainlog.
Zumindest für den Usecase eines Stromzählers mit optischer Schnittstelle ist für viele Zählertypen nicht möglich, die Werte der einzelne Phasen abzufragen.
Grundsätzlich ist das auch nicht gesetzlich erforderlich. Die einzelne WB nach KfW-Kritierien darf auch "dumm" einphasig oder dreiphasig laden.
OpenWB bietet m.W. drei "Netzsicherheitsfeatures":
1) Netzfrequenz-Auswertung
2) Schieflastbeachtung
3) Maximaler Strom je Phase
Es gibt nun verschiedene Lösungsmöglichkeiten:
a) Ist keine Phasenstrom-Info vorhanden, werden Fehler im main.log geschrieben und ein Status angezeigt
b) Die Beachtung des maximalen Phasenstroms lässt sich - wie die Schieflastbeachtung - abschalten. 
Vorschlag für das Sollverhalten:
Die 3 o.a. Kriterien werden alle einzeln auswählbar. Sind 2) und/oder 3) aktiv, ist das Fehler der Netzströme ein Fehler. Sind 2) und 3) nicht aktiv, erfolgt keine Fehlermeldung.

LKuemmel, 2022-02-01:
Guter Punkt. Ich werde es so überarbeiten, dass bei fehlenden Strömen eine Warnung angezeigt wird, dass das LM nur mit der max Leistung arbeitet.

Aktuell ist es so umgesetzt, dass fehlende Ströme automatisch auf 0 gesetzt werden [https://github.com/openWB/core/blob/3bddabc115406f536408198c999f87c6d925a065/packages/modules/common/component_state.py#L26](url)
Ich bin verwundert, dass bei Dir Exceptions geworfen werden. Verwendest Du das MQTT-Modul?

gvzdus, 2022-02-01:
Ja, ich verwende MQTT. Wie o.a.: Es gibt diverse Zähler, die keine phasenweisen Werte (Strom oder Leistung) ausgeben. Mein Ideal wäre also, dass sie lediglich dann die entsprechenden Netzschutzfunktionen nicht nutzen können.
Ich hatte übrigens einen Patch vorbereitet, aber richtig gut gefällt er mir nicht.

LKuemmel, 2022-02-01:
Ich habe auch schon eine Idee.
Magst mal herzeigen, alternative Lösungsansätze sind immer gut.

gvzdus, 2022-02-01:
Bitteschön: PR#285

LKuemmel, 2022-02-02:
Ich habe auf Basis deines PR den PR #287 erstellt.

closed 2022-02-03 by LKuemmel