openWB/core #329: Konfiguration MeterId in einer Fronius Komponente

vuffiraa72, 2022-02-28:
Beim Anlegen eines Fronius Systems wird die MeterId eines Smartmeter oder S0-Zählers abgefragt. Unabhängig davon, dass diese Konfiguration noch direkt im Zähler eingebunden werden muss, scheint die Prüfung des Wertes auf 0 oder 1 falsch.

Laut Fronius Solar API Kapitel 3.8 – GetMeterRealtimeData request ist die MeterId/DeviceId als positive Zahl definiert: 0..65535.

Bei meiner Installation hat der Smartmeter z.B. aufgrund einiger anfänglicher Probleme/Tests mit S0-Zählern letztendlich die ID 2 bekommen.

Ich schlage vor, die Werteprüfung entsprechend anzupassen.

benderl, 2022-03-01:
Hast Du einen Link zu der Doku? Vielleicht stehen da noch andere interessante Dinge drin.

vuffiraa72, 2022-03-02:
Die Doku findet man direkt bei Fronius: https://www.fronius.com/de/solarenergie/installateure-partner/downloads
Speziell https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2012.pdf

PR #351 (2022-03-22): update settings
fixes #329
closed 2022-03-22 by benderl