openWB/core #362: Hauptseite - Werte neben den Schiebereglern werden nicht dargestellt

MartinRinas, 2022-03-31:
In den Einstellungen zum PV Lademodus sind die Schieberegler für min Stromstärke, SoC etc. auf den konfigurierten Wert eingestellt, jedoch wird der Wert daneben nicht dargestellt sondern nur '--'.

[image]

MartinRinas, 2022-03-31:
Wenn ich den Schieberegler einmal verstelle taucht der Wert auf.

MartinRinas, 2022-04-10:
möglicherweise die gleiche Ursache für ein weiteres UI Problem auf der Hauptseite. 
Ladepunkt ist gesperrt, wenn ich später die Seite neu aufrufe ist der LP weiterhin gesperrt (Hinweistext), allerdings zeigt der Schalter das nicht an. Wenn ich dann sperre & entsperre passt es wieder.

[image]

benderl, 2022-05-11:
Solle jetzt wieder passen.

MartinRinas, 2022-05-11:
jup, in der Tat! danke :)

[image]