openWB/core #408: SOC Module HowTo Token Store

DetMoerk, 2022-05-19:
Hallo zusammen,
für diverse SoC Files müssen ja Tokens nach der Authentifizierung abgespeichert werden, um mit diesen Token dann den entsprechenden API Request zu machen. Gibt es schon einen Plan, wie und wo diese Token gespeichert werden sollen?
In den Broker, oder als File? An welcher Stelle im Filesystem sollen die liegen?

VG
Det

MartinRinas, 2022-09-29:
gibt nun einen funktionierenden Tesla SoC: https://github.com/openWB/core/blob/master/packages/modules/tesla/soc.py

benderl, 2022-09-30:
Funktionieren tut der, aber mit der Speicherung der sensiblen Daten bin ich noch nicht zufrieden. Aktuell ist die komplette Konfiguration inklusive Zugangsdaten im Broker und kann dort potentiell ausgelesen werden.
Wenn jemand dazu Ideen hat, gerne her damit. Das Thema "Benutzerverwaltung" kommt da auch noch ins Spiel.

DetMoerk, 2022-09-30:
Ganz ehrlich hilft mir das jetzt nicht direkt. dass es ein Telsa Modul gibt. Ich kenne keine Strukturen, wie ich die notwendigen Parameter wo ablegen soll, damit es dann auch so funktioniert, wie openWB sich das vorstellt! Es muss ja eine Verbindung zum Fahrzeug geben. Das kann und will ich mir nicht aus der Nase ziehen oder über endlose Tests dann ausprobieren. Ich hab auch keinen Tesla, um mir das dann dort ab zu schauen, wie das dann zu konfigurieren wäre, damit es am Fahrzeug hängt.
Und wenn es aktuell noch Bedenken mit der Speicherung der Anmeldedaten gibt, dann macht das für mich auch noch keinen Sinn.

benderl, 2022-09-30:
Aktuell verlangt das auch niemand. ;-)
Wenn die API soweit steht, gibt es weitere Infos.
Bei Ideen zur Speicherung der Zugangsdaten darf sich natürlich jeder vorab melden.

DetMoerk, 2022-12-29:
Als Erweiterung für die SoC Schnittstelle stelle ich mir noch etwas vor, um Fahrzeugindividuelle Daten mit abzuspeichern. Was auf jeden Fall bereits einmal gefordert wurde, um es auch im Ladelog protokollieren zu können sind der aktuelle km Stand. Aktuell sind ja SoC und Range enthalten. Die Mercedes API könnte das liefern. Als weiteren Vorschlag könnte ich mir auch vorstellen, eine Payload belibiger Form (JSON) zu ermöglichen, in die weitere Daten eingestellt werden können. Ich denke dabei z.B. auch an Daten die für eine Vorklimatisierung aus der WB genutzt werden können. Das liefert Mercedes aktuell glaube ich auch noch nicht, aber sowas könnte ich mir vorstellen. In der App eingebbar sind solche Schedules für die vorklimatisierung bereits.

VG
Det

DetMoerk, 2023-02-16:
@benderl; @LKuemmel: Gibt es eine Meinung von euch zu den Individuellen Daten für ein Fahrzeug?

VG
Det

DetMoerk, 2023-10-16:
Nothing more to say

DetMoerk, 2023-10-16:
Nothing more to say here, so closing.