openWB/core #424: JSON Modul Probleme

DetMoerk, 2022-05-29:
2022-05-29 14:36:17,075 - {soc.modules.common.fault_state:37} - ERROR - Json Zähler: FaultState FaultStateLevel.ERROR, FaultStr <class 'AttributeError'> 'Device' object has no attribute 'domain', Traceback:
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/json/device.py", line 66, in update
    response = req.get_http_session().get(self.domain, timeout=5)
AttributeError: 'Device' object has no attribute 'domain'

2022-05-29 14:36:17,081 - {soc.modules.common.fault_state:37} - ERROR - Json Wechselrichter: FaultState FaultStateLevel.ERROR, FaultStr <class 'AttributeError'> 'Device' object has no attribute 'domain', Traceback:
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/json/device.py", line 66, in update
    response = req.get_http_session().get(self.domain, timeout=5)
AttributeError: 'Device' object has no attribute 'domain'

Vermute hier müssen die Anpassungen aus openWB 1.9 auch noch nachgezogen werden. 
VG
Det

DetMoerk, 2022-05-29:
habe mir gerade die JSON Module aus der Nightly 1.9 auf die V2.0 geholt. Damit funktioniert es wieder :-). Der PR aus 1.9 sollte hier also auch funktionieren.

benderl, 2022-05-30:
Ja, das wird noch gemerged.

LKuemmel, 2022-05-31:
#426 ist nun gemergt.