openWB/core #640: Zeitladen findet sich nicht im Ladeprotokoll?

MartinRinas, 2022-11-14:
Ich nutz' Zeitladen um nachts das Auto auf den notwendigen Ladestand zu bringen. Hier ist mir heute aufgefallen dass die Ladevorgänge vom Zeitladen nicht im Ladeprotokoll zu finden sind, kann das sein?

Z.b. der Ladevorgang von heute Nacht (14.11.):
[image]

Im Ladeprotokoll ist nichts zu finden. Das Auto ist immer noch angesteckt, das Zeitfenster ist aber bereits verstrichen (1:00-6:30).  Auch von früheren Ladevorgängen ist nichts zu finden, jedoch wurde ein nächtlicher Ladevorgang wohl dem PV Laden zugeschlagen. Ich bekomm' definitiv keine 26kWh vom 6.11. 19:37 bis 7.11. 10:27 von meinen ~7kWp auf dem Dach. 
[image]

MartinRinas, 2022-11-15:
danke Lena. Update ist installiert, morgen früh kann ich berichten.

LKuemmel, 2022-11-15:
Ich habe den PR jetzt erst gemergt, habe noch auf das Review von Lutz gewartet. Also bitte nochmal ein Update machen. Vorher Backup erstellen nicht vergessen!

MartinRinas, 2022-11-16:
Hm, das scheint leider noch nicht geklappt zu haben. Laufe auf dieser Version: 2022-11-15 16:51:23 +0100 [f4e28462]
Auto ist seit gestern Mittag angesteckt. Lademodus PV, Zeitplan nachts aktiv. Tagsüber wurden ein paar wenige kWh über PV geladen. Nachts dann nochmal zusätzliche Energie über den Zeitplan. Im Ladelog ist dennoch beides als PV aufgeführt:
[image]

MartinRinas, 2022-11-17:
Mit den letzten Anpassungen im UI taucht der Ladevorgang nun im Protokoll auf. Leider scheint jeder Regelzyklus als eigenständiger Ladevorgang erfasst zu werden. gut 1200 Seiten an Einträgen für den November ;)

[image]

LKuemmel, 2022-11-21:
Mit #661 hat es bei mir nun funktioniert. Bitte nochmal testen.

MartinRinas, 2022-11-22:
danke @LKuemmel  das passt nun, es wird nur mehr ein Eintrag erstellt.

[image]

Es taucht nun zwar noch ein kurzer PV-Laden Eintrag auf, aber das ist inhaltlich sogar richtig da der Ladevorgang mit erreichen des SoC nicht unmittelbar gestoppt wird. Der Submode wird auf Stop und der Ladestrom auf 0, die Abschaltung wird aber dennoch überprungen da ein Ladevorgang aktiv ist die Abschaltschwellen erst noch durchlaufen werden. 

Hat aber nichts mit dem Logging zu tun, das funktioniert einwandfrei und zeigt die tatsächlichen Ladevorgänge inhaltlich korrekt. Lt. Statusmeldung läuft der Algorithmus in diese Verzweigung: https://github.com/openWB/core/blob/master/packages/control/pv.py#L196

LKuemmel, 2022-11-22:
Danke für die Rückmeldung

closed 2022-11-22 by LKuemmel