openWB/core #658: SoC wird beim Anstecken nicht aktualisiert

MartinRinas, 2022-11-18:
Kürzlich wurde ja das aktualisieren des SoC beim Anstecken nachgerüstet. Leider klappt das nicht immer zuverlässig, mal gehts und mal nicht, bin mir noch nicht sicher woran das hängt.

Ich hab' das eben mal reproduziert und einen Logauschnitt erfasst. 
[main.log](https://github.com/openWB/core/files/10043330/main.log)

LKuemmel, 2022-11-24:
#662 ist gemergt, bitte testen.

MartinRinas, 2022-11-24:
@LKuemmel ich hab' eben mal aktualisiert, nun scheint die Regelung aber nicht zu starten, es tut sich nach der 'boot done' Meldung nichts mehr im Log.

`Requirement already satisfied: packaging in /usr/local/lib/python3.9/dist-packages (from pytest==6.2.5->-r /var/www/html/openWB/requirements.txt (line 4)) (21.3)
Requirement already satisfied: iniconfig in /usr/local/lib/python3.9/dist-packages (from pytest==6.2.5->-r /var/www/html/openWB/requirements.txt (line 4)) (1.1.1)
Requirement already satisfied: requests<3,>=2.3 in /usr/lib/python3/dist-packages (from requests_mock==1.9.3->-r /var/www/html/openWB/requirements.txt (line 5)) (2.25.1)
Requirement already satisfied: idna>=2.0 in /usr/lib/python3/dist-packages (from yarl<2.0,>=1.0->aiohttp==3.7.2->-r /var/www/html/openWB/requirements.txt (line 7)) (2.10)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /usr/local/lib/python3.9/dist-packages (from packaging->pytest==6.2.5->-r /var/www/html/openWB/requirements.txt (line 4)) (3.0.6)
Installing collected packages: schedule
Successfully installed schedule-1.1.0
load versions...
chgrp: cannot access '/var/www/html/openWB/data/restore/*': No such file or directory
chmod: cannot access '/var/www/html/openWB/data/restore/*': No such file or directory
2022-11-24 13:29:05: boot done :-)`

MartinRinas, 2022-11-24:
Der Service läuft nicht, suche mal weiter..

`● openwb2.service - "Regelung openWB 2.0"
     Loaded: loaded (/var/www/html/openWB/data/config/openwb2.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-11-24 13:28:47 CET; 4min 25s ago
    Process: 521 ExecStart=/var/www/html/openWB/packages/main.py (code=exited, status=1/FAILURE)
   Main PID: 521 (code=exited, status=1/FAILURE)
        CPU: 197ms

Nov 24 13:28:47 openwb2 systemd[1]: openwb2.service: Scheduled restart job, restart counter is at 5.
Nov 24 13:28:47 openwb2 systemd[1]: Stopped "Regelung openWB 2.0".
Nov 24 13:28:47 openwb2 systemd[1]: openwb2.service: Start request repeated too quickly.
Nov 24 13:28:47 openwb2 systemd[1]: openwb2.service: Failed with result 'exit-code'.
Nov 24 13:28:47 openwb2 systemd[1]: Failed to start "Regelung openWB 2.0`

MartinRinas, 2022-11-24:
Dienst läuft nun nach systemctl restart openwb2

Auch nach einem neustart des OS bleibt der Dienst aktiv. Dann hat das womöglich mit dem Update nichts zu tun solange andere nicht das gleiche berichten.

Ich beobachte den SoC nach dem Anstecken und werde berichten wie sich das verhält.

MartinRinas, 2022-11-27:
Hat bisher zuverlässig funktioniert, ich schließe hier mal.

closed 2022-11-27 by MartinRinas