openWB/core #663: Daten der Monatsausertung deutlich zu gering

MartinRinas, 2022-11-21:
Ich hab' mir eben mal die Monatsauswertung für November angesehen, da stimmt etwas mit dem Graph nicht. Die Monatssummen kommen hin, rd. 300kWh habe ich an einem Ladepunkt bereits geladen.

Allerdings sehe ich diese Energiemengen überhaupt gar nicht im Chart:
[image]

So sieht der Monat im Sunnyportal aus:
[image]

Andere Monate sind ähnlich falsch, im Oktober geht die Skala gleich nur bis 2,8kWh:
[image]

Kann euch gern die Rohdaten, CSVs oder Logs liefern wenn etwas benötigt wird.

MartinRinas, 2022-11-22:
Ja, die Zählerstände sind korrekt, eben mal den EVU Zähler mit den Daten aus dem Sunnyportal abgeglichen, das stimmt gut. Die Daten im Broker sind auch korrekt und identisch mit denen aus dem CSV. Auffällig ist lediglich dass ich ein paar Einträge für den 1. November habe. Ich hab' die Daten mal in Excel gezogen und den Tagesimport für Zähler 7 (EVU) ausgerechnet:
[image]

benderl, 2022-11-22:
Kannst Du die JSON Datei bitte mal anhängen?

MartinRinas, 2022-11-22:
ja klar. (Musste die Dateien in .txt umbenennen, GitHub mag wohl keine .json Dateien)
[202211.json.txt](https://github.com/openWB/core/files/10071032/202211.json.txt)

[202210.json.txt](https://github.com/openWB/core/files/10071031/202210.json.txt)

benderl, 2022-11-23:
Mit PR #666 sollte die Darstellung jetzt passen.

MartinRinas, 2022-11-23:
ja in der Tat, danke @benderl, passt jetzt. 
[image]

closed 2022-11-23 by MartinRinas