openWB/core #726: Ladung wird nach Erreichen des Zeitladen-Limits nicht sofort gestoppt

MartinRinas, 2023-01-02:
Situation: Es ist ein Zeitladefenster mit Limit (z.b. SoC) konfiguriert, Lademodus PV mit 1p/3p Umschaltung. Zeitfenster ist aktiv, Ladevorgang wird gestartet. Kein PV Überschuss vorhanden.

Problem: Nach Erreichen des konfigurierten Limits (SoC) wird der Ladevorgang nicht sofort beendet. Viel mehr wird der Ladestrom auf 6A reduziert und weiter geladen. Lt. Statusmeldung wird der Ladevorgang nach erreichen der Abschaltschwelle von 180s gestoppt. Tatsächlich wird jedoch bis zum Erreichen der 1p/3p Umschaltung gewartet, die Umschaltung von 3p auf 1p durchgeführt, Ladevorgang weiterhin mit 6A. Jetzt greift die Abschaltschwelle und der Ladevorgang wird dann nach 180s beendet.

Erwartetes verhalten: Konfiguriertes Limit wird erreicht, da kein Überschuss vorhanden wir der Ladevorgang sofort beendet.

[image]
[main.log](https://github.com/openWB/core/files/10333562/main.log)

LKuemmel, 2023-01-03:
Das Limit muss auch im aktiven Lademodus eingestellt werden, damit in diesem Modus ebenfalls nicht weitergeladen wird. Unter Ladeprofile->PV-Laden das SoC-Limit wie in Zeitladen einstellen.

LKuemmel, 2023-01-03:
Die Lademodi aus 1.9 wurden etwas angepasst.

Für Deinen Anwendungsfall ist Zielladen gedacht: Es wird mit möglichst viel Überschuss der eingestellte SoC zur vorgegebenen Uhrzeit geladen und dann ggf bis zum SoC-Limit weitergeladen, wenn Überschuss vorhanden ist. Andernfalls die Ladung gestoppt.

MartinRinas, 2023-01-03:
Ja, das hatten wir an anderere Stelle schon einmal diskutiert. Ich halte Zeitladen hier unterm Strich für die bessere Option.
Zum einen ist Zeitladen einfacher in der Anwendung (Ich muss mir als Anwender keine Gedanken über die Akkugröße, Ladeverluste, maximale Ladleistung und andere technische Details meines Autos machen), zudem kann ich relativ einfach unterschiedliche Limits setzen. Zeitladen nachts geht bis 70%, das Limit fürs PV Laden ist deutlich höher. 

Ich stell' die Situation gerade noch mal im Simulator nach, ich meine ich hatte die entsprechende Stelle schon mal im Code gefunden. Vielleicht hab' ich ja eine Idee wie man diesen Übergang anders gestalten kann, bzw. diesen Fall adressieren kann.

MartinRinas, 2023-01-26:
Vielen Dank für den Fix 👍

closed 2023-01-26 by MartinRinas