openWB/core #777: Hausverbrauch wird falsch berechnet

vuffiraa72, 2023-01-25:
Passend zu meinem Eintrag im Forum https://openwb.de/forum/viewtopic.php?p=79349#p79349 habe ich mal geschaut, voran es liegen könnte.

Das Problem scheint die Berechnung in `CounterAll.calc_daily_yield_home_consumption()` in counter_all.py zu sein:
```
            if len(data.data.pv_data) > 1:
                pv = data.data.pv_all_data.data.get.daily_exported
            else:
                pv = 0
            if len(data.data.bat_data) > 1:
                bat_imported = data.data.bat_all_data.data.get.daily_imported
                bat_exported = data.data.bat_all_data.data.get.daily_exported
            else:
                bat_imported = 0
                bat_exported = 0
            if len(data.data.cp_data) > 1:
                cp_imported = data.data.cp_all_data.data.get.daily_imported
                cp_exported = data.data.cp_all_data.data.get.daily_exported
            else:
                cp_imported, cp_exported = 0, 0
```
Hier wird immer geprüft, ob es mehr als einen Element in den Dictionaries gibt. Für PV ist dort z.B. bei mir nur der eine konkrete Eintrag: `{'pv0': <control.pv.Pv object at 0xffff4be8ab50>}`.

Kann es sein, dass hier der Eintrag `all` wegoptimiert wurde?
Wenn  meine Annahme stimmt, müssten die Prüfungen auf `> 0` geändert werden.

Einen Verweis auf `all` habe ich so weit nur noch in `measurement_log_test.py` gefunden. Das passt zu meiner Vermutung, ist ja "nur" ein (alter) Test.

Weiterhin gibt es einige Stellen, wo speziell geprüft wird, ob die Keys in den Dictionaries nur mit den Komponentenprefixen beginnen. Wenn es `all` nicht mehr gibt, braucht man das auch nicht mehr prüfen, z.B. in pv_all.py `PvAll.calc_power_for_all_components()`.

Wenn mir jemand meine Annahmen bestätigt, kann ich auch einen passenden PR erstellen.

benderl, 2023-01-30:
Sollte mit #783 erledigt sein.

vuffiraa72, 2023-01-30:
Die Werte passen jetzt bei mir.
Für mich ist das Problem damit behoben.
Danke für die schnelle Bearbeitung!

closed 2023-01-30 by vuffiraa72