openWB/core #815: Automatische Phasenumschaltung wird nicht ausgeführt

vuffiraa72, 2023-02-12:
Ich hatte das Verhalten auch schon kurz um Forum beschrieben: https://openwb.de/forum/viewtopic.php?p=79999#p79999

Im Log kann man aber ganz gut nachvollziehen, was wirklich passiert. Das konnte ich auch so mit dem Simulator nachstellen und debuggen

Zuerst scheint alles zu funktionieren:
```
2023-02-07 10:33:11,164 - {control.chargepoint:701} - {DEBUG:MainThread} - EV-Phasenzahl beschränkt die nutzbaren Phasen auf 3
2023-02-07 10:33:11,167 - {control.chargepoint:692} - {DEBUG:MainThread} - Phasenzahl Lademodus: 1
2023-02-07 10:33:11,169 - {control.chargepoint:867} - {DEBUG:MainThread} - LP 2, EV: Standard-Fahrzeug (EV-Nr.0): Theoretisch benötigter Strom 6A, Lademodus pv_charging, Submodus: Chargemode.PV_CHARGING, Phasen: 1, Priorität: False, max. Ist-Strom: 15.73

2023-02-07 10:33:11,192 - {control.algorithm.algorithm:26} - {DEBUG:MainThread} - # Algorithmus
2023-02-07 10:33:11,193 - {control.ev:517} - {DEBUG:MainThread} - Phasenumschaltung kann nun durchgeführt werden.
2023-02-07 10:33:11,210 - {control.process:22} - {DEBUG:MainThread} - # Ladung starten.
2023-02-07 10:33:11,212 - {control.chargelog:52} - {DEBUG:MainThread} - imported_since_mode_switch 4468.989999999991 counter 221468.99
2023-02-07 10:33:11,220 - {control.phase_switch:37} - {DEBUG:MainThread} - Thread zur Phasenumschaltung an LP2 gestartet.
2023-02-07 10:33:11,222 - {control.chargepoint:623} - {DEBUG:MainThread} - start phase switch phases_to_use 1control_parameter phases 3
2023-02-07 10:33:11,223 - {control.chargepoint:354} - {INFO:MainThread} - LP 2: Umschaltung von 1 auf 3 Phasen.
2023-02-07 10:33:11,226 - {control.process:106} - {INFO:MainThread} - LP2: set current 0 A
```

In der nächsten Runde wird jetzt die Umschaltung vorbereitet:
```
2023-02-07 10:33:20,588 - {root:61} - {INFO:MainThread} - # ***Start*** 
2023-02-07 10:33:20,963 - {control.chargepoint:701} - {DEBUG:MainThread} - EV-Phasenzahl beschränkt die nutzbaren Phasen auf 3
2023-02-07 10:33:20,964 - {control.chargepoint:689} - {DEBUG:MainThread} - Umschaltung wird durchgeführt, Phasenzahl nicht ändern 3
2023-02-07 10:33:20,964 - {control.chargepoint:692} - {DEBUG:MainThread} - Phasenzahl Lademodus: 3
2023-02-07 10:33:20,967 - {control.chargepoint:867} - {DEBUG:MainThread} - LP 2, EV: Standard-Fahrzeug (EV-Nr.0): Theoretisch benötigter Strom 6A, Lademodus pv_charging, Submodus: Chargemode.PV_CHARGING, Phasen: 3, Priorität: False, max. Ist-Strom: 0.0

2023-02-07 10:33:20,987 - {control.algorithm.algorithm:26} - {DEBUG:MainThread} - # Algorithmus
2023-02-07 10:33:21,002 - {control.process:22} - {DEBUG:MainThread} - # Ladung starten.
2023-02-07 10:33:21,010 - {control.chargepoint:582} - {DEBUG:MainThread} - phase switch running
2023-02-07 10:33:21,011 - {control.process:106} - {INFO:MainThread} - LP2: set current 0 A
```
Nach dem Log `phase switch running` wird in Chargepoint.initiate_phase_switch() `charging_ev.data.control_parameter.timestamp_perform_phase_switch = None` gesetzt. Soweit ich das verstehe, schließt das die Phasenumschaltung für die Logik ab. Die Ladung läuft aber noch nicht. Daher wird dann beim nächsten Intervall neu geprüft:
```
2023-02-07 10:33:30,401 - {root:61} - {INFO:MainThread} - # ***Start*** 
2023-02-07 10:33:30,713 - {control.chargepoint:701} - {DEBUG:MainThread} - EV-Phasenzahl beschränkt die nutzbaren Phasen auf 3
2023-02-07 10:33:30,715 - {control.chargepoint:692} - {DEBUG:MainThread} - Phasenzahl Lademodus: 1
2023-02-07 10:33:30,717 - {control.chargepoint:867} - {DEBUG:MainThread} - LP 2, EV: Standard-Fahrzeug (EV-Nr.0): Theoretisch benötigter Strom 6A, Lademodus pv_charging, Submodus: Chargemode.PV_CHARGING, Phasen: 1, Priorität: False, max. Ist-Strom: 0.0

2023-02-07 10:33:30,741 - {control.algorithm.algorithm:26} - {DEBUG:MainThread} - # Algorithmus
2023-02-07 10:33:30,751 - {control.chargepoint:354} - {INFO:MainThread} - LP 2: Die Ladung wird gestartet, sobald nach 30s die Einschaltverzögerung abgelaufen ist. 
2023-02-07 10:33:30,759 - {control.process:22} - {DEBUG:MainThread} - # Ladung starten.
2023-02-07 10:33:30,762 - {soc.modules.common.component_context:24} - {DEBUG:cp2} - Update Komponente ['Ladepunkt']
2023-02-07 10:33:30,764 - {control.phase_switch:37} - {DEBUG:MainThread} - Thread zur Phasenumschaltung an LP2 gestartet.
2023-02-07 10:33:30,766 - {control.chargepoint:623} - {DEBUG:MainThread} - start phase switch phases_to_use 3control_parameter phases 1
2023-02-07 10:33:30,767 - {control.chargepoint:354} - {INFO:MainThread} - LP 2: Umschaltung von 3 auf 1 Phase.
2023-02-07 10:33:30,769 - {control.process:106} - {INFO:MainThread} - LP2: set current 0 A
```
Der Algorithmus entscheidet jetzt aber `start phase switch phases_to_use 3control_parameter phases 1`. Das passiert in Chargepoint.get_phases_by_selected_chargemode(). Die Logik sieht keine aktive Umschaltung mehr, es wird aber (noch) nicht geladen. Daher wird geprüft:
```
(not charging_ev.ev_template.data.prevent_phase_switch or
 self.data.set.log.imported_since_plugged == 0) and
 self.data.config.auto_phase_switch_hw)
```
Die Umschaltung ist nicht verboten, es wurde aber bereits (einphasig) geladen und die passende HW ist vorhanden, also wird hier hart auf einphasig gestellt. Diese Bedingungen funktionieren beim Zurückschalten auf einphasiges Laden bzw. wenn das Laden überhaupt beginnt. Beim Umschalten auf dreiphasiges Laden muss aber die eingestellte Phasenzahl beibehalten werden.

Leider sehe ich hier nicht, wie man erkennen könnte, dass die Phasenumschaltung erst noch wirklich abgeschlossen werden muss.

LKuemmel, 2023-02-17:
Wenn nach der Umschaltung wieder Strom freigegeben wird, startet der Skoda innerhalb von 10s die Ladung (Ladestatus true/Stecker grün)?

vuffiraa72, 2023-02-20:
Ja, das dauert keine 3 Sekunden.
Also im UI erscheint die Freigabe 0W / 6A, dann klackt es im Auto und gleich danach ist Stecker und auch die Anzeige am Auto direkt grün.

LKuemmel, 2023-03-06:
Das Problem war, dass die Einschaltverzögerung gestartet wurde. Ist behoben und die Umschaltung von 3P auf 1P funktioniert hier.

LKuemmel, 2023-04-13:
Die Phasenumschaltung wurde mittlerweile komplett überarbeitet.

closed 2023-04-13 by LKuemmel