openWB/core #833: GitHub Desktop kann nicht mehr Clonen

hhoefling, 2023-03-02:
```
Tutorial: Erstkonfiguration.md
Tutorial: Konfiguration Geräte und Komponenten.md
```
Seit die beiden Tutorial-Dateien im DOCS Zweig sind kann man das Repro nicht mehr
mit GitHub Desktop clonen.
Github Desktop stolper über den ":" im Namen .
Nach Entfernen der Dateien klappt das Clonen mit GitHub Desktop wieder.
(Probiert in eigenem Fork)

benderl, 2023-03-02:
Windows?

hhoefling, 2023-03-02:
Jup.

DetMoerk, 2023-03-03:
Ist bei mir auch so. Habe auch Github Desktop. Warte dann einfach auf den PR

DetMoerk, 2023-03-03:
@LKuemmel :Funktioniert im master wieder. für die angelegten Feature Branches müsste man das auch noch nachziehen

hhoefling, 2023-03-03:
Ok. Löschen ist auch ein Weg.
Aber einfach nur den ":" aus dem Namen nehmen hätte es auch getan..
( oder falls umbennen nicht geht, per Cut&Past in eine neue Seite verpflanzen.)

benderl, 2023-03-03:
> @LKuemmel :Funktioniert im master wieder. für die angelegten Feature Branches müsste man das auch noch nachziehen

Rebase der Branches schiebe ich nach, wenn der PR #836 gemerged wurde. Eigentlich kann der master noch nicht funktionieren. ;-)

benderl, 2023-03-07:
Bitte testen, PR ist im master.

closed 2023-03-08 by benderl