openWB/core #878: HTTP Counter modules.common.simcount._calculate <class 'TypeError'> '<' not supported between instances of 'float' and 'NoneType'

nidr, 2023-04-29:
Hello,

I tried to add a http Counter to grab the SDM630 from my ioBroker via simple api. 
However, in the openwb logs i find the correct value (466.85 in this case) the http counter should show, but it comes with an error which prevents the value from beeing shown in the webinterface. 

> 2023-04-29 14:30:50,076 - {urllib3.connectionpool:452} - {DEBUG:device8} - http://192.168.40.6:8087 "GET /getPlainValue/modbus.1.inputRegisters.30053_Total_system_power HTTP/1.1" 200 6
2023-04-29 14:30:50,078 - {modules.common.req:11} - {DEBUG:device8} - Get-Response: 466.85
2023-04-29 14:30:50,078 - {modules.common.simcount._simcount:38} - {DEBUG:device8} - Previous state: SimCounterState(timestamp=1682769020.4472444, power=None, imported=0, exported=0)
2023-04-29 14:30:50,083 - {modules.common.fault_state:40} - {ERROR:device8} - HTTP Zähler: FaultState FaultStateLevel.ERROR, FaultStr modules.common.simcount._calculate <class 'TypeError'> '<' not supported between instances of 'float' and 'NoneType', Traceback: 
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/fault_state.py", line 99, in wrapper
    return delegate(*args, **kwargs)
  File "/var/www/html/openWB/packages/modules/common/simcount/_calculate.py", line 14, in calculate_import_export
    power_low = min(power1, power2)
TypeError: '<' not supported between instances of 'float' and 'NoneType'

> The above exception was the direct cause of the following exception:

> Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/configurable_device.py", line 24, in __call__
    self.__updater(component)
  File "/var/www/html/openWB/packages/modules/devices/http/device.py", line 37, in <lambda>
    component_updater=IndependentComponentUpdater(lambda component: component.update(session))
  File "/var/www/html/openWB/packages/modules/devices/http/counter.py", line 38, in update
    imported, exported = self.sim_counter.sim_count(power)

nidr, 2023-04-29:
It seems that it is not sufficient to provide the current power in the configuration.
Please add a rule or a check to the webinterface so that any really needed information must be provided by user.
After filling all fields in the http counter with adresses, the error does not appear anymore.

benderl, 2024-11-04:
Fixed in https://github.com/openWB/openwb-ui-settings/pull/548