openWB/core #998: Lastmanagement mit PV-Anlage

Bascht74, 2023-07-02:
Kann die Seite Lastmanagement bitte noch mit einem weiteren Beispiel mit PV-Anlage (und ggf. Speicher) erweitert werden?

Es geht um folgende Seite:
https://github.com/openWB/core/wiki/Lastmanagement-und-kaskadierte-Z%C3%A4hler

Aktuell abe ich meine PV-Anlage so konfiguriert: 
[image]

Ich bin mir aber nicht sicher, ob das so richtig ist und ich die PV-Anlage an der richtigen Stelle in der Struktur eingeklinkt habe.

Gruß und besten Dank für einen Tipp oder eine Aktualisierung der Seite (werden sicher noch mehr Personen mit dieser Frage existieren).

MartinRinas, 2023-07-17:
An welcher Stelle soll denn wie begrenzt werden? In deinem Setup werden die beiden Ladepunkte gemeinsam auf 22kW/32A begrenzt. Z.b. weil es sich um eine gemeinsame Zuleitung mit entsprechender Last handelt. Ob der Strom nun von der PV Anlage oder aus dem Netz kommt spielt keine Rolle, die beiden Ladepunkte dürfen gemeinsam nicht mehr als 22kW/32A abgeben.

Eine Begrenzung des Hausanschlusses wird direkt am 'MQTT-Zähler Strom Hauptzähler' konfiguriert, dem sog. EVU Zähler. Also dem Zähler an oberster Stelle in der Hierachie.

benderl, 2023-11-07:
Geschlossen, weil keine Rückmeldung mehr kam.

closed 2023-11-07 by benderl