openWB/core #1075: SoC wird nicht gespeichert, womöglich timeout

MartinRinas, 2023-08-04:
Tesla SoC wird zuverlässig abgerufen und taucht im Log auf, allerdings fehlt die letzte Meldung zum speichern des CarState wenn das Auto zuerst aufgeweckt werden muss.
Ist das Auto vor dem Abruf wach wird alles gespeichert.

https://github.com/openWB/core/pull/1029#issue-1799855544

Flock82, 2023-08-04:
Gleiches Problem bei Kia. Kann der SoC nicht innerhalb einiger Sekunden (evtl. vor der nächsten Regelschleife?) bereitgestellt werden, wird er nicht gespeichert.

LKuemmel, 2023-08-08:
Bitte mit dem Feature-Branch feature_timeout_soc testen. Vor dem Branchwechsel eine Sicherung erstellen.

MartinRinas, 2023-08-08:
jup, teste ich gleich mal.

Flock82, 2023-08-09:
Funktioniert bei mir auch mit den Änderungen

Flock82, 2023-08-21:
Leider hat der Fix eine unschöne Nebenwirkung, die mir erst jetzt aufgefallen ist: Während das SoC-Modul läuft, werden die Weete von den anderen Modulen(in meinem Fall alle MQTT) nicht mehr übernommen, was unschöne Lücken im Graphen erzeugt.

benderl, 2023-08-31:
Die Anpassung ist im master. Bitte einmal testen.

Flock82, 2023-09-04:
Läuft jetzt gut bei mir.

benderl, 2023-09-06:
Dann schließe ich mal.

closed 2023-09-06 by benderl