openWB/core #1096: Editing manual SOC > 0% on main web page gives inconsistent results

andlem74, 2023-08-11:
After the user has set an initial SOC (e. g. 15%) on the main web page, the displayed SOC will increase as charging progresses, e. g. up to 45%. At any time, the user can edit the SOC again. The pop-up dialog will show the current value, e. g. 45%, and the user will be able to decrease or increase it with the + and - buttons, e. g. to 48% as this is what the car reports. Afterwards, the value the user entered will not be used to change the current value, but the initial one the user had entered earlier, i. e. the initial SOC of 15% will be replaced. Hence, the displayed value will change to 48%-15%+45%-15%=63%. This is both unexpected and confusing. The user will expect the estimated SOC to be updated to the value of 48% because that’s what he entered.

LKuemmel, 2023-10-18:
should be solved with #1143

benderl, 2023-11-07:
closed as no feedback provided

closed 2023-11-07 by benderl