openWB/core #1129: Manueller SoC - wird auf den Wert vor dem letzten Ladevorgang zurückgesetzt, sobald am nächsten Tag PV-Ertrag wieder startet

VanillaFord, 2023-09-17:
SW 2.1.1 beta2, Series 2 DUO

Manueller SoC: ...siehe auch: https://openwb.de/forum/viewtopic.php?p=92745#p92745

Der SoC springt auf den Wert vor der letzten Ladung zurück...die Ladung ist im Ladelog zu sehen....laut Timeline dann, wenn die PV wieder Erzeugung bringt (der PV-Ertrag steht im UI des "nachts" übrigens auf "-0")
Es passiert aber nur bei dem LP, bei dem auch am Vortag mit PV geladen wurde...am LP2 - gestern Auto auch angesteckt und SOC eingestellt, aber nicht geladen - blieb der SOC konstant.

benderl, 2023-09-17:
Sie #1097

closed 2023-09-17 by benderl

VanillaFord, 2023-09-17:
In diesem Fall , anders als in [#1097](https://github.com/openWB/core/issues/1097)  wurde weder Zielladen benutzt, noch das Auto auf 100% geladen, sondern die Ladung manuell gestoppt.
Der "Trigger" für den Reset scheint hier der Neustart der PV am nächsten Tag zu sein (wenn am LP zuvor geladen wurde)....das ist reproduzierbar.