openWB/core #1146: SyntaxError in data.py, annotated name can't be global

uis0815, 2023-10-01:
Tried to get openWB from git-master running on raspi3 with debian buster (python version 3.7.3). Unfortunately, there seems to be a problem in data.py:

Okt 01 21:37:52 raspberrypi systemd[1]: Started "Regelung openWB 2.0".
Okt 01 21:37:52 raspberrypi main.py[12325]: Traceback (most recent call last):
Okt 01 21:37:52 raspberrypi main.py[12325]:   File "/var/www/html/openWB/packages/main.py", line 12, in <module>
Okt 01 21:37:52 raspberrypi main.py[12325]:     from helpermodules.measurement_logging.update_daily_yields import update_daily_yields
Okt 01 21:37:52 raspberrypi main.py[12325]:   File "/var/www/html/openWB/packages/helpermodules/measurement_logging/update_daily_yields.py", line 4, in <module
Okt 01 21:37:52 raspberrypi main.py[12325]:     from control import data
Okt 01 21:37:52 raspberrypi main.py[12325]:   File "/var/www/html/openWB/packages/control/data.py", line 514
Okt 01 21:37:52 raspberrypi main.py[12325]:     data: Data
Okt 01 21:37:52 raspberrypi main.py[12325]:     ^
Okt 01 21:37:52 raspberrypi main.py[12325]: SyntaxError: annotated name 'data' can't be global

benderl, 2023-10-02:
Please use debian Bullseye for installation.

benderl, 2023-10-19:
Closed because of missing feedback.

closed 2023-10-19 by benderl