openWB/core #1165: enable WiFi

toyota-axel, 2023-10-12:
enable WiFi / WLAN again.
Default to wired Ethernet (and DHCP), have WiFi as sub-menu for specialists.

Saharel001, 2023-11-06:
@toyota-axel I think this i not a good idea because modbus or mqtt is not so stable over wifi. Maybe there are timeout issues or other problems and the customer calls openWB support.

toyota-axel, 2023-11-06:
Yes but, having WiFi at least easily accessible & configurable / enabled would allow for people to test and decide on their own instead of goin through pains of laying Ethernet cable to the outdoors.

dibadodo, 2024-02-03:
I can only agree with Axel. Switching off WIFI is not always helpful. If someone then connects an external AP or Powerline in front of it, the latency may be even higher. However, troubleshooting is much more difficult in this case because you can't see the connection. With the local WIFI adapter, I can assess the quality of the connection and even display a message or similar if necessary.
That's why I'm clearly calling for WIFI to be reactivated.

closed 2026-04-15 by kevinwieland