openWB/core #1186: SMB Backup: Dateien werde nicht abgelegt

Saharel001, 2023-10-20:
Hallo zusammen, dass samba Modul scheint noch nicht so ganz zu funktionieren.
Ich habe einen Share auf einen QNAP angegeben, User, Passwort. Leider bleibt das Share leer.

Probiert habe ich: 
//IP/Share/Folder
//IP/Share
\\IP\Share\Folder
\\IP\Share

Leider fehlt hier ein Wikieintrag zu.

Auf der openwb unter http://IP/openWB/data/backup/ wird immer eine frische Datei erzeugt.
Da das Modul wenig bis keine logs ausspuckt, kann ich es nur beschreiben.

Die Verbindung via Explorer ist kein Problem, Schreibrechte hat der User auch:
[image]

Saharel001, 2023-10-20:
Ich habe den Fehler gefunden, in der Initialisierung der SMB Verbindung ist der Name 'server' hard gecoded ;) Außerdem wurde nicht nach https://pysmb.readthedocs.io/en/latest/api/smb_SMBConnection.html gearbeitet.

Werde es beheben und nen PR stellen.

Saharel001, 2023-10-20:
PR steht: https://github.com/openWB/core/pull/1187

Saharel001, 2023-11-07:
durch PR erledigt

closed 2023-11-07 by Saharel001

Sleepwalker86, 2024-11-17:
Danke für deine Korrektur, das war mein erster Versuch mit ein wenig Hilfe von ChatGPT. :-D Bin leider kein Programmierer. Vielen Dank nochmal für die Korrektur.
Gruß Sascha