openWB/core #1188: PSA SoC wird abgerufen und gespeichert, ist aber in der UI nicht zu sehen.

MartinRinas, 2023-10-21:
Ich hab' jetzt mal das PSA Modul konfiguriert um die manuelle Berechnung testen zu können.

PSA Abruf passt, SoC und Reichweite wird gespeichert.

`2023-10-21 16:33:43,747 - {modules.vehicles.psa.api:120} - {INFO:fetch soc_ev15} - psa.fetch_soc: soc=48%, range=112, timestamp=2023-10-21T13:16:17Z
2023-10-21 16:33:43,767 - {modules.common.configurable_vehicle:61} - {DEBUG:fetch soc_ev15} - Requested start soc from api: CarState(soc=48, range=112, soc_timestamp=2023-10-21T13:16:17Z)%
2023-10-21 16:33:43,774 - {modules.common.store._api:26} - {DEBUG:fetch soc_ev15} - Raw data CarState(soc=CarState(soc=48, range=112, soc_timestamp=2023-10-21T13:16:17Z), range=None, soc_timestamp=None)
2023-10-21 16:33:43,803 - {modules.common.store._api:30} - {INFO:store soc_ev15} - Saving CarState(soc=CarState(soc=48, range=112, soc_timestamp=2023-10-21T13:16:17Z), range=None, soc_timestamp=None)`

Tauchen im Broker auf - aber nur im calculated_soc_state, nicht im soc vom Fahrzeug selbst:
[image]

Und in der UI sehe ich auch keinen SoC:
[image]

MartinRinas, 2023-10-23:
jup, ist nun besser, danke!

Reichweite fehlt noch, wobei ich nicht weiß ob die zuvor übernommen wurde. Aus der API wird die Reichweite geliefert:

`2023-10-23 15:19:39,569 - {modules.vehicles.psa.api:120} - {INFO:fetch soc_ev15} - psa.fetch_soc: soc=83%, range=224, timestamp=2023-10-23T12:44:56Z
2023-10-23 15:19:39,588 - {modules.common.configurable_vehicle:61} - {DEBUG:fetch soc_ev15} - Requested start soc from api: 83%
2023-10-23 15:19:39,595 - {modules.common.store._api:26} - {DEBUG:fetch soc_ev15} - Raw data CarState(soc=83, range=None, soc_timestamp=None)
2023-10-23 15:19:39,626 - {modules.common.store._api:30} - {INFO:store soc_ev15} - Saving CarState(soc=83, range=None, soc_timestamp=None)`

[image]

MartinRinas, 2023-10-23:
Wobei sich die Frage stellt was man mit der Reichweite während manueller Berechnung macht. Solange die Daten durch die jeweilige SoC API abgerufen werden ist das okay, während des Ladevorgangs stimmt die Reichweite dann nicht mehr.

Nasdero, 2023-10-24:
Hmm komisch, heute wird es richtig angezeigt, hatte gestern sogar die WB neu gestartet und den Corsa ab und an gesteckt...

closed 2023-10-26 by LKuemmel