openWB/core #1201: Ausgabe der Werte von Kostal Plenticore inkonsistent

DerHerrW, 2023-10-24:
Für eine sinnvolle Berechnung der Kostal-Werte muß die Batterie parallel zum Hybrid-WR in der Struktur angelegt werden (entgegen der Anleitung im Wiki). Andernfalls wird der Hausverbrauch genau wie die PV-Leistung um die Batterieladeleistung erhöht.

Die hat jedoch zur Folge, daß bei Entladung der Batterie eine PV-Leistung angezeigt wird, die nicht vorhanden ist (hier in der Nacht):
[image]
Die Entladeleistung der Batterie findet sich als PV-Leistung des Wechselrichters.

Seltsamerweise stimmt die Darstellung im Hauptfenster in dieser Konfiguration:
[image]

Wenn ich den Speicher verschachtelt unterhalb des Wechselrichters anordne, wie das Wiki es für Hybridwechselrichter angibt, wird die Speicherladung auf die AC-Ausgabe des WRs aufgeschlagen, das Ergebnis als PV-Ertrag ausgegeben und der angezeigte Hausverbrauch erhöht sich um den Wert der Speicherladung (Bild umfaßt die Umkonfiguration):
[image]

Ich habe bemerkt, daß es eine rege Diskussion zum Plenticore gibt:
https://github.com/snaptec/openWB/pull/2440#discussion_r996275286

DerHerrW, 2023-10-26:
Ich habe etwas weiter gesucht. Wenn die Sonne untergegangen ist und die DC-Leistung 0 sein sollte und das Haus aus der Batterie versorgt werden soll, ist die DC-Leistung anscheinend nicht exakt Null, sondern bei mir leicht negativ (vmtl wegen Meßtoleranzen). Nur, wenn der Hausverbrauch unter meine Standbyschwelle fällt und der WR in Standby geht, ist die Leistung exakt Null:
[image]

in device.py im Kostal-Plenticore-Verzeichnis ist ein Pfad davon abhängig, ob die DC-Leistungen in Summe größer oder gleich 0 sind:
[image]

Wenn die DC-Leistungssumme =0 ist (Standby), wird der Codepfad des oberen Pfeil ausgeführt. Die Leistungen Batterie/Hausverbruach passen.
Wenn die Summe der DC-Leistungen aber kleiner Null ist, wird der untere Pfad genutzt. Ich vermute, daß hier der Grund ist, warum im Auswertediagramm nachts bei Batterieentladung PV-Leistung bei mir angezeigt wird:

[image]

Ich habe die Vermutung, daß die Langzeitanzeige hier eine PV-Erzeugung errechnet

DerHerrW, 2023-10-26:
Wenn ich etwas ausprobieren soll, bräuchte ich ssh-Zugang zu meiner OpenWB oder eine Anleitung, wie ich den Code auf meinem Desktop-PC unter Linux laufen lassen kann, damit die Wallbox als secondary definiert werden kann.

DerHerrW, 2023-10-28:
Wenn ich die Tageslogs aus der Wallbox hole (z.B. http://192.168.1.102/openWB/data/daily_log/20231026.json), sehe ich, daß im Item PV der Zähler des Plenticores in Wh abgelegt ist. Dieser zählt allerdings die abgegebene AC-Energie.
Wenn der Akku z.B: nachts entlädt, zählt der PV-Zähler des Plenticores weiter, da er die Akku-Energie abgibt und man bekommt eine nicht vorhandene PV-Erzeugung angezeigt.
[image]
Vermutlich wäre hier sinnvoll, den Speicher als Untergerät vom Wechselrichter anzulegen, allerdings ist das Plenticore-Module so wie ich es verstehe so programmiert, daß bei den Leistungen für PV tatsächlich die gesamte PV incl Batterieladung herauskommt und nicht nur die AC-Ausgangsleistung.

Die eingangs im Issue beschriebene Inkonsistenz scheint durch die Berechnung des 120min-Diagrammes (Nutzt das Leistungen?) und der Tagesdiagramme (Nutzen Zähler) zu entstehen.

Eine erste Idee wäre, das Plenticore-Modul so zu überarbeiten, daß als PV-Leistung nur die abgegebene AC-Leistung ausgegeben wird. Momentan: (DC-Leistungen der Strings, korrigiert um die Wandlungsverluste). Dann könnte der Speicher des Plenticore unterhalb des Wechselrichters angordnet werden wie im Wiki beschrieben.

DerHerrW, 2023-10-30:
Ich habe mir jetzt ssh-Zuggang verschafft und in einem ersten Schuß die unterschiedliche Berechnung von pv_state durch ein einheitliches
`pv_state = InverterState(power=inverter_state.power,  exported=inverter_state.exported)`
ersetzt.
Die Struktur habe ich so geändert wie im Wiki beschrieben, also Anordnung des Speichers als Element des Hybridwechselrichters:
[image]
Das Ergebnis scheint mir erstmal plausibel. Das Laden hat auch funktioniert und die leistungsbasierte Kurzzeitanzeige wie auch das Tageslog scheinen zumindest bis gerade eben plausibel:
[image]
[image]
Mit Abfall der DC-Leistungs auf <=0 hat sich wohl noch etwas verändert. Meine aktuelle Änderung:
[image]
Ich werde als nächstes den Fall dc_in<0 ebenfalls vereinheitlichen.

Frage ist, wie man mit einem Update kommuniziert, daß die Struktur angepaßt werden muß - bisher musste der Speicher auf gleicher Höhe wie der Wechselrichter sein, damit das Laden fehlerfrei tat.

DerHerrW, 2023-10-31:
So, nun habe ich ohne Berücksichtigung der Verluste einfach in device.py in jedem Fall pv_state=inverter_state gesetzt, nun sieht es auch beim abendlichen Akkuentlehren plausibel aus, wenn der Akku ein Untergerät des Wechselrichter in der Struktur ist:
[image]

Vorläufiger Code in /var/www/html/openWB/packages/modules/devices/kostal_plenticore/device.py:
[image]

Um die verschwundenen Watt (die noch nicht im Hausverbrauch auftauchen), muss ich mich später kümmern.
Benötigt jemand einen Pullrequest? Ich würde vermutlich mehrere Anläufe brauchen, weil ich das noch nie gemacht habe.

DerHerrW, 2023-11-02:
Pullrequest #1231 ist erstellt. Mein allererster PR :-)

checklarsb, 2023-11-12:
Genau das Problem habe ich auch. Bin gespannt auf die Änderung. Vielen Dank für deine Arbeit.

DBU73, 2023-11-17:
Ich habe dieses Problem auch. Plenticore mit Speicher und KSEM als EVU Modul im Netzanschluss-Zweig. Ich musste den Speicher neben (nicht hinter) dem Wechselrichter anordnen, aber beim nächtlichen Speicherentladen wurde die Speicherentladung zusätzlich als PV-Leistung gezählt. Der Plenticore ist auf der aktuellen Firmware und mit der openWB Version 1.9.303 tritt dieses Problem nicht auf.

checklarsb, 2023-11-28:
Weiß jemand, warum der PR nicht freigegeben wird?

DerHerrW, 2023-12-06:
Grundsätzlich scheint es zu funktionieren. Heute nacht allerdings hat mein Akku eine Notladung hingelegt, und die Ladung hat falscherweise PV-Arbeit erzeugt. Ich werde noch untersuchen, was da zu tun ist, sofern nicht jemand anders die Lösung schon kennt.

[image]

Die relevanten Einträge sind 68 und 69:
[20231206_notladung_pvleistung.json](https://github.com/openWB/core/files/13590307/20231206_notladung_pvleistung.json)

[image]

Der Zähler des Plenticore scheint das unter exported als abgegebene Leistung verbucht zu haben? Ich kann mir heute keinen Kopf mehr darum machen, werde aber darüber nachdenken. Wenn Kostal da was falsch verbucht, kann man vermutlich nichts dran machen.

DerHerrW, 2023-12-06:
Kurz nachgedacht: Ist kein Problem der Leistung, deshalb schließe ich diesen Issue, denn die Leistung passt soweit ich sehe:

[image]

Falls ich für die Energie eine Lösung habe, mache ich einen neuen PR auf.

checklarsb, 2023-12-07:
Ok, das kannst es dir ja nochmal durch den Kopf gehen lassen, wie man das mit der Zwangsladung ändert. Hast du dann diesen PR zurückgezogen? Hätte ihn gerne mal getestet. Gruß Lars