openWB/core #1202: Mögliche Division durch 0 in core/packages/modules/devices/kostal_plenticore /device.py

DerHerrW, 2023-10-24:
Ich hoffe, ich verstehe das richtig:

In core/packages/modules/devices/kostal_plenticore /device.py findet sich Zeile 46-50:
`                        # Wird die Batterie entladen, werden die Wandlungsverluste anteilig an der DC-Leistung auf PV
                        # und Batterie verteilt. Dazu muss der Divisor Total_DC_power != 0 sein.
                        power_gross = dc_in - bat_state.power
                        pv_state = InverterState(power=dc_in / power_gross * inverter_state.power,
                                                 exported=inverter_state.exported)`

Wenn dc_in (Die Summer der vom WR gelieferten DC-Leistungen) größer 0 ist und exakt gleich groß wie bat_state.power (Ladeleistung der Batterie), kommt es zur Division durch 0 in Zeile 49.

Dieser Zustand legt einen Hausverbrauch von 0 nahe, den man in der Tat auch erreichen kann, wenn man noch ein Balkonkraftwerk am Start hat.

benderl, 2023-10-25:
Ich denke, dieser Sonderfall ist schon sehr konstruiert und unwahrscheinlich. Trotzdem kann es natürlich dazu kommen.
Bitte gerne einen PR mit einer Anpassung stellen.

Saharel001, 2023-10-25:
Wenn der SMB PR durch ist kann ich mir das anschauen.

DerHerrW, 2023-10-26:
Ich glaube, ich hatte einen Denkfehler: Die Division kann nur dann kommen, wenn DC-Leistung>=0 ist und die Batterie entladen wird: dc_in ist >=0 und bat_state.power<0. Damit kann der Divisor power_gross = dc_in - bat_state.power nur größer werden als Null, da (-bat_state.power) sicher >0 ist.

closed 2023-10-26 by DerHerrW