openWB/core #1215: Status View

Saharel001, 2023-10-27:
Ist kein Bug nur ein Vorschlag. 
Um das Display Thema konsistent zu halten, wäre es gut die Statusview auch in eine dash-board-card zu packen.
Würde dann so aussehen:
[image]

@benderl  @LKuemmel @MartinRinas 
Was haltet ihr davon?

MartinRinas, 2023-10-27:
einheitliches Look&Feel finde ich immer gut! 
Kann man bei der Statusseite das Label und den Wert in die gleiche Zeile packen? also

IP-Adresse: <Wert>
anstelle

IP-Adresse:
<Wert>

Wirkt m.E. dann aufgeräumter und der vorhandene Platz wird besser genutzt.

Saharel001, 2023-10-27:
> einheitliches Look&Feel finde ich immer gut! Kann man bei der Statusseite das Label und den Wert in die gleiche Zeile packen? also
> 
> IP-Adresse: 192.168.178.57 anstelle
> 
> IP-Adresse: 192.168.178.57
> 
> Wirkt m.E. dann aufgeräumter und der vorhandene Platz wird besser genutzt.

Du meinst so?
[image]

MartinRinas, 2023-10-27:
ja, da finde ich persönlich schon deutlich besser. Vielleicht die Boxen noch etwas breiter machen damit die nach links rutschen, dann passt auch die Versionsnummer komplett rein.

benderl, 2023-10-27:
Mit dem Layout des Status bin ich soweit einverstanden. Müsste noch geprüft werden, wie das dann auf verschiedenen Anzeigen skalieren kann. Das sollte aber das Framework hergeben.

Die "Gesamtladung des Ladepunktes" soll genau was beinhalten? Den aktuellen Zählerstand oder die Energie des laufenden Ladepunktes? Beides würde zumindest von den Daten her zum Ladepunkt gehören und müsste im linken Bereich integriert werden. Rechts sind ausschließlich Daten vom Fahrzeug.

Saharel001, 2023-10-27:
> Mit dem Layout des Status bin ich soweit einverstanden. Müsste noch geprüft werden, wie das dann auf verschiedenen Anzeigen skalieren kann. Das sollte aber das Framework hergeben.
>
> Die "Gesamtladung [...]

Hier geht es darum den Gesamtverbrauch (import) des LP im Display sehen zu können.

benderl, 2023-10-27:
Also den Zählerstand? Was ist hier der Anwendungsfall? Die geladene Energie für diesen Ladevorgang ist mir ja noch plausibel.

Saharel001, 2023-10-27:
> Also den Zählerstand? Was ist hier der Anwendungsfall? Die geladene Energie für diesen Ladevorgang ist mir ja noch plausibel.

In vielen Unternehmen ist die Abrechnung über Fotos vom Zählerstand möglich. Vor der Ladung und am Ende je ein Bild.
Für mich als Besitzer irrelevant da ich das Ladeprotokoll habe, wenn ich aber Besuch habe und dieser möchte Laden und die Kosten bei seinem AG abrechnen, dann schon.

Wenn das zu speziell ist, kein Thema Im Forum gibt es dazu aber auch anfragen.

benderl, 2023-10-27:
Ich würde es vorerst zurückstellen.

Saharel001, 2023-10-27:
Alle klar, habe nur die Statusview mal als draft PR gemacht. https://github.com/openWB/core/pull/1217

closed 2023-11-13 by Saharel001