openWB/core #1219: Ständige timeout Fehler bei EVCC Openwb 2.0

atze09, 2023-10-29:
Nach Umstellung von openwb 1.9 auf 2.0 bekomme ich nun ständig timeout Fehler bei EVCC und die Anbindung funktioniert nicht . An der Topologie wurde nichts verändert.
EVCC läuft auf Home Assistant und die openwb ist über modbustcp angebunden

Ein Ping ist unauffällig.
Softwarestand ist 2023-10-27 14:04:58 +0200 [8778d1eed] auf dem master branch.

benderl, 2023-10-29:
Du müsstest schon ein paar mehr Infos geben. Die Konstellation mit HA über Modbus ist definitiv nicht der Standard. Was sagen die Logmeldungen? Und bitte nicht nur die zwei Zeilen aus dem Issue in 1.9.

atze09, 2023-10-29:
Bei mir läuft EVCC auf dem HA Server und regelt meine goe perfekt.
Nachdem ich nun auf openwb Software 2 gewechselt bin, da diese nun von EVCC supportet wird, funktioniert die Anbindung der openwb leider nicht mehr. Die Anbindung der openwb in EVCC ist nun per ModbusTCP.
Der folgende Logoutput kommt leider ständig:
2023-10-29 16:07:19,784 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc, Payload None darf nicht 'None' sein.
2023-10-29 16:07:19,789 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc_timestamp, Payload None darf nicht 'None' sein.
2023-10-29 16:07:21,632 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc, Payload None darf nicht 'None' sein.
2023-10-29 16:07:21,638 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc_timestamp, Payload None darf nicht 'None' sein.
2023-10-29 16:07:23,441 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc, Payload None darf nicht 'None' sein.
2023-10-29 16:07:23,446 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc_timestamp, Payload None darf nicht 'None' sein.
2023-10-29 16:07:25,277 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc, Payload None darf nicht 'None' sein.
2023-10-29 16:07:25,283 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc_timestamp, Payload None darf nicht 'None' sein.
2023-10-29 16:07:27,110 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc, Payload None darf nicht 'None' sein.
2023-10-29 16:07:27,116 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc_timestamp, Payload None darf nicht 'None' sein.
2023-10-29 16:07:28,903 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc, Payload None darf nicht 'None' sein.
2023-10-29 16:07:28,908 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc_timestamp, Payload None darf nicht 'None' sein.
Log des internen Ladepunktes

Ansonsten hab ich die log Datei durchsucht aber finde nichts bzgl. Modbus.
Kann gerne noch weitere Info's geben, weiß aber gerade nicht was.

atze09, 2023-10-29:
Mit dem Programm Modbus Master unter Windows bekomme ich auch einen Zugriff. Muss ich bei der Openwb noch irgendwas einstellen.
Den secondary Modus habe ich bereits gewählt.
Das sind die Modbus Einstellungen seitens EVCC:
  modbus: tcpip
  id: 1
  host: 192.168.17.80 # Hostname
  port: 1502 # Port

benderl, 2023-10-29:
Die beiden Meldungen sagen lediglich, dass ungültige Daten an die openWB gesendet werden. Das kann von EVCC über Modbus kommen, oder aber auch direkt vom HA. Mehr kann man daraus nicht ableiten.

Das sieht für mich eher nach einem individuellen Problem aus. Da bist Du vermutlich im Forum besser aufgehoben, da dort HA auch häufig genutzt wird.

atze09, 2023-10-29:
Super danke, dann probiere ich es übers Forum.

atze09, 2023-10-29:
Die Meldungen kommen von der noch aktivierten Integration die aber noch nicht auf 1.9 upgedated ist.

atze09, 2023-10-29:
Der zweite Haken für Morbus aktivieren,unterhalb der Wahl secondary war nicht aktiv. Das war das Problem, vielen Dank für den Support!

closed 2023-10-29 by atze09