openWB/core #1224: ModbusTCP Register 10160 RFID Fehler

atze09, 2023-10-30:
Hallo,

Ich habe eine Problem mit dem RFID Register des Modbus TCP.
Alle anderen Register erreiche ich via Windows Modbus Master. Jedoch bei 10160 und den folgenden bekomme ich einen Fehler.
[image]

LKuemmel, 2023-11-02:
Ich kenne das Programm nicht... Muss bei Start Address nicht die 10160 eingetragen werden?

atze09, 2023-11-02:
[image]
Stimmt das hatte ich

> Ich kenne das Programm nicht... Muss bei Start Address nicht die 10160 eingetragen werden?

[image]

Stimmt das hatte ich eigentlich auch drin, hier nochmal der richtige Screenshot mit gleichem Fehler.
Ich habe es nur Parallel mit dem Programm probiert, da EVCC auch einen Fehler bei der Verbindung auswirft. Wollte wissen, ob der Fehler aufseiten EVCC oder openwb liegt.
Scheinbar passt hier was seitens openwb nicht?
Betrifft auch nur dieses Register alle anderen sind verfügbar.
Echt komisch

LKuemmel, 2023-11-03:
Hast Du einen Tag gescannt?

atze09, 2023-11-03:
Nein ich habe keinen RFID reader.

benderl, 2023-11-14:
Bitte mit dem aktuellen Master testen. Es gab einen Fix für nicht vorhandene RFID Tags.

closed 2023-12-11 by benderl