openWB/core #1263: Huawei Modbus Error after install Beta for Tronity

danielrudolf, 2023-11-18:
After updating to 2.0, the Huawei setup works perfectly and everything works. But since there is no Tronity there, I installed the beta. Tronity would be available here, but it gives an error during the Modbus query for Huawei. Then I can set what I want or set it to the current release. The Huawei query ends in a Modbus error. Only when I delete the SD card and install the 2.0 again can I set up Huawei again and it works. And the game starts all over again. Is there already a solution for this?

Modulmeldung:
modules.common.modbus <class 'modules.common.fault_state.FaultState'> ('modules.common.modbus Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response', <FaultStateLevel.ERROR: 2>)

danielrudolf, 2023-11-19:
OK, unfortunately I have to correct it. This error also occurs after a few minutes in the release version. Why can't you take over the plugin 1:1 from 1.9? Everything would work perfectly. Only the daily target loading is missing there.

LKuemmel, 2023-11-21:
Please post logs in details mode

benderl, 2023-12-04:
Closed due to no response.

closed 2023-12-04 by benderl