openWB/core #1298: Option/Alternative zu "Ladung aktiv halten" ähnlich zu 1.9

miradarya, 2023-12-10:
Die derzeitige Implementierung der Funktion "Ladung aktiv halten" hält einen einmal freigegebenen Ladevorgang für immer aktiv für Fahrzeuge, die nach einer Ladepause nicht wieder aufwachen. Es gibt Fahrzeuge (PSA-Fahrzeuge mit bestimmten HW/SW-Versionen des OBC), die ein anderes Problem haben. Bei diesen wird die Ladeleistung nach einer Ladefreigabe verzögert und nur sehr langsam hochgefahren. Wenn währenddessen die Ladefreigabe wieder entzogen wird, interpretiert der OBC dies als Ladestörung und ein Ladevorgang kann erst wieder stattfinden, nachdem das Ladekabel aus- und wieder eingestöpselt wurde (die CP-Unterbrechung hilft in diesem speziellen Fall nicht).

Es muss daher bei diesen Fahrzeugen sichergestellt sein, dass eine einmal erteilte Ladefreigabe für einen gewissen Zeitraum aktiv gehalten wird, um dem OBC genug Zeit zum Hochfahren zu geben. Hierfür gab es in der 1.9 von @MartinRinas eine identisch bezeichnete Option "Ladung aktiv halten" mit einem solchen einstellbaren Timeout. Dort war diese Funktion allerdings nur bei CP-Unterbrechung aktiv, was nicht bei allen Fahrzeugen geholfen hat.

Es wäre deshalb gut, wenn es für die Funktion "Ladung aktiv halten" in der 2.x einen entsprechen einstellbaren Timeout gäbe. Für Fahrzeuge mit der Aufwachproblematik (bisherige Funktionsweise) wäre der Timeout quasi unendlich, für die erwähnten PSA-Fahrzeuge z.B. 30 Sekunden.

Siehe auch [https://openwb.de/forum/viewtopic.php?p=90747](https://openwb.de/forum/viewtopic.php?p=90747)

MartinRinas, 2023-12-21:
Ich hab' mir das eben mal angeschaut, leider ist mir nicht klar wo ich das einbauen kann. Der Algorithmus ist in der 2.0er doch deutlich komplexer geworden, das erschließt sich mir nicht mehr so richtig.

Gebraucht wird m.E. folgendes:
* Konfigurationsoption im Fahrezugprofil die Ladung nach Freigabe mindestens x Sekunden aktiv zu halten. (das bekomme ich hin, kein Problem)
* Zeitstempel bei der Erteilung einer Ladefreigabe setzen (Soll-Ladestrom steigt von 0A auf >0A)
* Vor Entzug der Ladefreigabe aufgrund von Nicht-Laden prüfen ob der ^^ Zeitstempel länger abgelaufen ist als konfiguriert. Falls ja, Ladestrom auf 0 setzen. Falls nein, nichts machen, Ladestrom unverändert lassen.

Das würde es unseren PSA Modellen erlauben aufzuwachen und mit der Ladung zu beginnen. Nach erteilter Ladestromfreigabe brauchen die rd. 20-30s bis tatsächlich geladen wird. Entfällt in dieser Zeit die Ladefreigabe (z.b. weil kein PV Überschuss mehr vorhanden ist), geht das Auto leider auf Störung und verweigert jegliche Ladung bis abgesteckt wurde.

miradarya, 2024-08-21:
Update: Seit dieses Jahr genug Sonne zum PV-Laden vorhanden ist, hatte ich den oben beschriebenen Fall mit der Ladestörung nicht ein einziges Mal, letztes Jahr im Herbst war das mehrfach aufgetreten. Vielleicht gab es in der Zwischenzeit in der Firmware eine entscheidende Änderung, aber für den Moment bin ich mir ziemlich sicher, dass das Problem nicht mehr besteht. Wenn sich nicht noch jemand anderes mit demselben Fehlerbild meldet, schließe ich hier erstmal.

closed 2024-08-21 by miradarya