openWB/core #1314: Tagesdiagramm in aktuellstem Build wirft Fehler

DetMoerk, 2023-12-20:
Bei der Verarbeitung des Befehls 'getDailyLog' mit den Parametern '[object Object]' ist ein Fehler aufgetreten:
Es ist ein interner Fehler aufgetreten: Traceback (most recent call last):
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 126, in on_message
    func(connection_id, payload)
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 560, in getDailyLog
    get_daily_log(payload["data"]["day"]))
  File "/var/www/html/openWB/packages/helpermodules/measurement_logging/process_log.py", line 142, in get_daily_log
    data = _collect_daily_log_data(date)
  File "/var/www/html/openWB/packages/helpermodules/measurement_logging/process_log.py", line 153, in _collect_daily_log_data
    next_date = timecheck.get_relative_date_string(date, day_offset=1)
AttributeError: module 'helpermodules.timecheck' has no attribute 'get_relative_date_string'

DetMoerk, 2023-12-20:
Verhalten tritt auch bei Monthly Log auf.

Bei der Verarbeitung des Befehls 'getMonthlyLog' mit den Parametern '[object Object]' ist ein Fehler aufgetreten:
Es ist ein interner Fehler aufgetreten: Traceback (most recent call last):
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 126, in on_message
    func(connection_id, payload)
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 564, in getMonthlyLog
    get_monthly_log(payload["data"]["month"]))
  File "/var/www/html/openWB/packages/helpermodules/measurement_logging/process_log.py", line 167, in get_monthly_log
    data = _collect_monthly_log_data(date)
  File "/var/www/html/openWB/packages/helpermodules/measurement_logging/process_log.py", line 178, in _collect_monthly_log_data
    next_date = timecheck.get_relative_date_string(date, month_offset=1)
AttributeError: module 'helpermodules.timecheck' has no attribute 'get_relative_date_string'

benderl, 2023-12-20:
Sollte durch #1318 behoben sein.

DetMoerk, 2023-12-20:
Habe gerade aktualisiert. Sieht wieder gut aus. Diagramme werden angezeigt.

closed 2023-12-20 by DetMoerk