openWB/core #1349: Wrong permissions for GPIO access to internal chargepoint

markl65536, 2024-01-12:
After installing core 2.1.2-Patch 1 on a new SD card, I noticed that switching charging to 3 phases failed and wallbox kept charging with 1 phase only. Upgrading to 2.1.2-Patch 2 did not solve the issue.

Logs for internal chargepoint showed an exception in `internal_chargepoint_handler.py`, method `init_gpio` with the message *No access to /dev/mem. Try running as root!* and I suspected that this is the root cause.

Manually adding user `openwb` to groups `gpio` and `kmem` fixed the issue (`usermod -a -G gpio,kmem openwb`) and wallbox is working as expected now.

I do not know if this is the correct fix or if something else is going on. Maybe setting these group memberships should be included in `openwb-install.sh`?

For OS installation, I used image `2023-12-05-raspios-bullseye-armhf-lite`.

LKuemmel, 2024-01-23:
behoben in #1365 
Bitte testen. Ein neues Image gibt es erst nach dem nächsten Release.
Nach der Installation des aktuellen Image ein Update auf den Master ausführen, dann sollte das Problem behoben sein.

benderl, 2024-05-03:
Wegen Inaktivität geschlossen.

closed 2024-05-03 by benderl