openWB/core #1378: Ladepunkt sperren nicht mehr möglich

andreasmuellerka, 2024-01-28:
Wann der Bug reingekommen ist, kann ich nicht genau sagen, aber der (Zeit-)Bereich ist klar:
Zwischen dem letzten stable Release: Release (2024-01-03 12:21:20 +0100 [06bcf54f9])
und dem aktuellen Master: 2024-01-25 10:50:43 +0100 [cb6674a2e]

Wenn ein Fahrzeug an einem Ladepunkt angeschlossen ist, kann man den Ladepunkt nicht mehr sperren. Klickt man im Webinterface auf "Sperren", wird der Schalter zwar kurz rot, schaltet sich aber wenige Sekunden später wieder auf grün (ungesperrt). Im Ergebnis wird der Ladepunkt nicht gesperrt, egal wie oft man es versucht.

Erwartetes Verhalten: Ladepunkt lässt sich auch sperren, wenn ein Fahrzeug angesteckt ist.

benderl, 2024-01-29:
Das Verhalten kann ich nicht nachstellen, daher scheint es kein generelles Problem zu sein. Du solltest Dich beim Support melden, wenn Du eine gekaufte openWB verwendest. Falls nicht, kann Dir eventuell im Forum geholfen werden. Da müssen dann aber auch aussagekräftige Logdateien gepostet werden.

closed 2024-01-29 by benderl

andreasmuellerka, 2024-01-29:
Gerne wende ich mich an den Support. Gleichwohl vermute ich stark, dass es ein neuer Bug ist, weil an der Konfiguration nichts verändert wurde. Es wurde lediglich auf den master geupdatet (siehe erster Post) und seitdem geht das Ladepunkt sperren nicht mehr. Das hat unmittelbar vor dem Update aber noch funktioniert.