openWB/core #1410: Nach RFID-Tag: Keine Änderung des Fahrzeugs oder Ladepunkt sperren mehr möglich

andreasmuellerka, 2024-02-08:
Voraussetzung: RFID-Tag ist einem Fahrzeug und Ladepunkt zugeordnet.

Ablauf: 1. Tag scannen, 2. Fahrzeug anstecken, 3. Versuchen Fahrzeugauswahl am Ladepunkt zu ändern oder versuchen den Ladepunkt zu sperren.

Fehlerhaftes Verhalten: Es lässt sich kein anderes Fahrzeug am Ladepunkt im Webinterface auswählen. Auch der Ladepunkt lässt sich nicht mehr sperren.

LKuemmel, 2024-02-09:
> Es lässt sich kein anderes Fahrzeug am Ladepunkt im Webinterface auswählen.
Was ist der Use Case? RFID dient doch gerade dazu, ein Fahrzeug zuzuordnen.
>  Auch der Ladepunkt lässt sich nicht mehr sperren.
Nur wenn Freigabe mit RFID aktiviert ist. Da ein Tag gescannt wurde, wird der Ladepunkt freigegeben.

andreasmuellerka, 2024-02-09:
> Was ist der Use Case? RFID dient doch gerade dazu, ein Fahrzeug zuzuordnen.
Ja, aber doch nur _einmalig_ nach dem Scannen des RFIDs. Warum das künstlich unterbinden?

> Nur wenn Freigabe mit RFID aktiviert ist. Da ein Tag gescannt wurde, wird der Ladepunkt freigegeben.

Ja klar, aber doch nur einmalig. Man sollte den Ladepunkt trotzdem danach jederzeit sperren können, um die Ladung zeitweise zu unterbrechen (ohne den Lademodus wechseln zu müssen, denn das erzeugt einen neuen Eintrag im Ladelog).
Auch hier dir Frage: Warum erfolgt die Freigabe und das Zuordnen via RFID nicht einmalig und danach verhält sich alles wie vorher auch. Warum künstlich das Sperren verhindern?!