openWB/core #1428: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte

Speedy1991, 2024-02-17:
After the last update to `2024-01-03 12:21:20 +0100 [06bcf54f9]` my wb is no more working - charging is never started. Digging in the logs I found this in the main log:

```
2024-02-17 13:53:00,969 - {control.prepare:36} - {ERROR:MainThread} - Fehler im Prepare-Modul
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/control/prepare.py", line 21, in setup_algorithm
    data.data.general_data.check_ripple_control_receiver()
  File "/var/www/html/openWB/packages/control/general.py", line 186, in check_ripple_control_receiver
    configured = hardware_configuration.get_hardware_configuration_setting(
  File "/var/www/html/openWB/packages/helpermodules/hardware_configuration.py", line 16, in get_hardware_configuration_setting
    return json.loads(f.read())[name]
  File "/usr/lib/python3.9/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
```
And this 

```
2024-02-17 13:52:51,228 - {modules.internal_chargepoint_handler.clients:132} - {DEBUG:Internal Chargepoint} - resolved_devices['/dev/ttyUSB0']
2024-02-17 13:52:51,229 - {modules.internal_chargepoint_handler.clients:143} - {ERROR:Internal Chargepoint} - LP0 Device: /dev/ttyUSB0
2024-02-17 13:52:51,383 - {modules.common.evse:57} - {DEBUG:Internal Chargepoint} - FW-Version: 18
2024-02-17 13:52:51,384 - {modules.internal_chargepoint_handler.clients:48} - {DEBUG:Internal Chargepoint} - <modules.common.modbus.ModbusSerialClient_ object at 0x715e6fd0>
2024-02-17 13:52:51,384 - {modules.internal_chargepoint_handler.clients:49} - {ERROR:Internal Chargepoint} - Modbus-ID der EVSE an LP0: 1
2024-02-17 13:52:52,396 - {modules.internal_chargepoint_handler.clients:66} - {DEBUG:Internal Chargepoint} - <modules.common.modbus.ModbusSerialClient_ object at 0x715e6fd0>
2024-02-17 13:52:52,397 - {modules.internal_chargepoint_handler.clients:67} - {DEBUG:Internal Chargepoint} - Zähler <class 'modules.common.mpm3pm.Mpm3pm'> mit Modbus-ID:5 antwortet nicht.
2024-02-17 13:52:52,454 - {modules.internal_chargepoint_handler.clients:63} - {ERROR:Internal Chargepoint} - Verbauter Zähler: <class 'modules.common.sdm.Sdm630'> mit Modbus-ID: 105
2024-02-17 13:52:52,614 - {modules.common.evse:57} - {DEBUG:Internal Chargepoint} - FW-Version: 18
2024-02-17 13:52:54,556 - {modules.common.fault_state:55} - {ERROR:Internal Chargepoint} - Interner Ladepunkt 0: FaultState FaultStateLevel.ERROR, FaultStr modules.common.modbus <class 'modules.common.fault_state.FaultState'> ('modules.common.modbus Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 2 bytes (0 received)', <FaultStateLevel.ERROR: 2>), Traceback: 
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/modbus.py", line 90, in __read_registers
    raise FaultState.error(__name__+" "+str(response))
modules.common.fault_state.FaultState: ('modules.common.modbus Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 2 bytes (0 received)', <FaultStateLevel.ERROR: 2>)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/internal_chargepoint_handler/internal_chargepoint_handler.py", line 154, in __init__
    self.cp0 = HandlerChargepoint(self.cp0_client_handler, 0, mode, global_data, parent_cp0, hierarchy_id_cp0)
  File "/var/www/html/openWB/packages/modules/internal_chargepoint_handler/internal_chargepoint_handler.py", line 221, in __init__
    self.module = chargepoint_module.ChargepointModule(
  File "/var/www/html/openWB/packages/modules/internal_chargepoint_handler/chargepoint_module.py", line 41, in __init__
    version = self.__client.evse_client.get_firmware_version()
  File "/var/www/html/openWB/packages/modules/common/evse.py", line 55, in get_firmware_version
    version = self.client.read_holding_registers(1005, ModbusDataType.UINT_16, unit=self.id)
  File "/var/www/html/openWB/packages/modules/common/modbus.py", line 123, in read_holding_registers
    return self.__read_registers(
  File "/var/www/html/openWB/packages/modules/common/modbus.py", line 105, in __read_registers
    raise FaultState.error(__name__+" "+str(type(e))+" " +
modules.common.fault_state.FaultState: ("modules.common.modbus <class 'modules.common.fault_state.FaultState'> ('modules.common.modbus Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 2 bytes (0 received)', <FaultStateLevel.ERROR: 2>)", <FaultStateLevel.ERROR: 2>)

2024-02-17 13:52:54,558 - {modules.internal_chargepoint_handler.internal_chargepoint_handler:296} - {ERROR:Internal Chargepoint} - Fehler im internem Ladepunkt
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/internal_chargepoint_handler/internal_chargepoint_handler.py", line 292, in handler
    self.internal_chargepoint_handler.loop()
  File "/var/www/html/openWB/packages/modules/internal_chargepoint_handler/internal_chargepoint_handler.py", line 193, in loop
    with SingleComponentUpdateContext(self.cp0.module.component_info):
AttributeError: 'InternalChargepointHandler' object has no attribute 'cp0'
```
I have a OpenWb Duo with a 3P/1P setup like this:
[image]
[image]

It looks like the cp0 (i guess this is charging point 0?) is no more found or mounted faulty. I didn't change anything on my openWb the last weeks and this error started after the upgrade to the `2024-01-03 12:21:20 +0100 [06bcf54f9]` release

I'm not sure if this issue should be in core or [openWb](https://github.com/snaptec/openWB/issues/2819) - feel free to remove or close one of them

Speedy1991, 2024-02-17:
Maybe this is a problem?
`Modbus-ID der EVSE an LP0: 1`
The LP ids are 2 and 3 - there is no one with ID 1

tpd-opitz, 2024-02-17:
> Maybe this is a problem? `Modbus-ID der EVSE an LP0: 1` The LP ids are 2 and 3 - there is no one with ID 1

When configuring an OWB - DUO this way it uses hard coded ModBus-IDs which are 1 (EVSE) and 5 (meter)  for LP1 an 2 (EVSE) and 6 (meter) for LP2.

However, the numbers you enter in the configuration form are not ModBus IDs but just a selector for the first or the second set of EVSE/meter pair.  Nothing wrong with the config at this point.

Nevertheless I'd recommend re-configuring the LPs (delete and create new).

benderl, 2024-02-19:
The first error is already fixed in current Beta and master branches.
Please try one of those (preferable master) after creating a current backup. Maybe the second error has already been solved, too.

benderl, 2024-05-21:
Closed due to inactivity.

closed 2024-05-21 by benderl