openWB/core #1504: Touch am display funktioniert mit V2.x nicht

nicomania, 2024-03-27:
Hallo,
ich dachte mir nun auch, ich wage den Umzug von 1.9 auf 2.1.3. Leider funktioniert direkt nach der Installation der Touch meines internen Displays nicht mehr. **_Also nur der Touch, die Anzeige wird korrekt dargestellt._** Alles andere funktioniert auch prima (Modbus etc.). Ich kann auch das Interface des Displays im PC-Browser aufrufen (http://<ip>/openWB/web/display/), bedienen und konfigurieren. Ich kann am Display auch per Touch bei schwarzem Display das Display aus dem standby „wecken“. Jedoch kann ich dann auf der Oberfläche herum drücken wie ich möchte, es rührt sich nichts.

Es handelt sich um einen OpenWB Eigenbau mit Raspi 3b und einem 7 Zoll DSI LCD von Waveshare mit 800x480 (https://www.waveshare.com/7inch-dsi-lcd.htm), angeschlossen per Flachbandkabel und sollte identisch zum originalen OpenWB Display sein. Es wird vom Hersteller als "driver free" mit i2c beschrieben, siehe Link. Unter Version 1.9 funktioniert es auch einwandfrei. Auch bei einem Rückumzug auf V1.9 funktioniert es wieder einwandfrei. Da es sich unter V2 ja auch aus dem standby wecken lässt, kann ich einen Hardwaredefekt eigentlich ausschließen. Ich tippe eher auf ein Konfigurationsproblem in einer Startdatei oder sowas.

Ich habe OpenWB 2.1.3 Release Patch 1 ausprobiert, sowie 2.1.2 und 2.1.0. Es macht in Bezug auf das Display keinen Unterschied. Ich verwende das Theme Cards und auch mit einem völlig neuen und cleanen image, nur mit Aktivierung des Displays und sonst keinerlei Konfiguration, tritt das Problem bereits auf.

Ich würde ja selbst etwas herum probieren, jedoch gibt es ja leider keinen SSH Zugriff mehr. Falls doch bitte ich um Hinweise.

Ich bitte um Hilfe. Sieht man was in der Log?
Danke und VG

```
rc.local bug not found
no factory reset pending, normal startup
atreboot.sh started
checking init in /boot/config.txt...
watchdog for atreboot.sh on pid 397 started, waiting for 900s
already up to date
Group membership...
Final group membership: openwb : openwb adm dialout cdrom sudo audio video plugdev games users input render netdev spi i2c gpio
Network...
Wait for connection...RTNETLINK answers: Network is unreachable
.RTNETLINK answers: Network is unreachable
.RTNETLINK answers: Network is unreachable
.RTNETLINK answers: Network is unreachable
.RTNETLINK answers: Network is unreachable
.RTNETLINK answers: Network is unreachable
.RTNETLINK answers: Network is unreachable
.
my primary IP: 192.168.178.99
my primary interface: eth0
running as primary, enabling plug'n'play network
apt configuration already up to date
install required packages with 'apt-get'...
OK:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
OK:2 http://archive.raspberrypi.org/debian bullseye InRelease
Paketlisten werden gelesen…
Paketlisten werden gelesen…
Abhängigkeitsbaum wird aufgebaut…
Statusinformationen werden eingelesen…
accountsservice ist schon die neueste Version (0.6.55-3).
apache2 ist schon die neueste Version (2.4.56-1~deb11u2).
bc ist schon die neueste Version (1.07.1-2+b1).
chromium ist schon die neueste Version (106.0.5249.119-1~deb11u1).
chromium-l10n ist schon die neueste Version (106.0.5249.119-1~deb11u1).
git ist schon die neueste Version (1:2.30.2-1+deb11u2).
jq ist schon die neueste Version (1.6-2.1).
libapache2-mod-php ist schon die neueste Version (2:7.4+76).
lightdm-autologin-greeter ist schon die neueste Version (1.0-3).
mmc-utils ist schon die neueste Version (0+git20180327.b4fe0c8c-1).
mosquitto ist schon die neueste Version (2.0.11-1+deb11u1).
mosquitto-clients ist schon die neueste Version (2.0.11-1+deb11u1).
openbox-lxde-session ist schon die neueste Version (0.99.2-4).
php ist schon die neueste Version (2:7.4+76).
php-curl ist schon die neueste Version (2:7.4+76).
php-gd ist schon die neueste Version (2:7.4+76).
php-json ist schon die neueste Version (2:7.4+76).
php-xml ist schon die neueste Version (2:7.4+76).
socat ist schon die neueste Version (1.7.4.1-3).
sshpass ist schon die neueste Version (1.09-1).
ssl-cert ist schon die neueste Version (1.1.0+nmu1).
sudo ist schon die neueste Version (1.9.5p2-3+deb11u1).
vim ist schon die neueste Version (2:8.2.2434-3+deb11u1).
x11-xserver-utils ist schon die neueste Version (7.7+8).
xserver-xorg ist schon die neueste Version (1:7.7+22).
lightdm ist schon die neueste Version (1.26.0-7+rpt1).
python3-pip ist schon die neueste Version (20.3.4-4+rpt1+deb11u1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 84 nicht aktualisiert.
done
[... 173 lines cut ...]
PvAllData(config=Config(configured=True), get=Get(daily_exported=0.0, fault_str='Kein Fehler.', fault_state=0, monthly_exported=0, yearly_exported=0, exported=0, power=0))
2024-03-26 20:17:40,571 - {control.data:266} - {INFO:MainThread} - system
{'update_in_progress': False, 'perform_update': False, 'ip_address': '192.168.178.99', 'current_commit': '2024-03-19 06:39:23 +0100 [4c209fd17]', 'current_branch': 'Release', 'current_branch_commit': '2024-03-19 06:56:08 +0100 [a6b648645]', 'current_missing_commits': ['2024-03-19 06:56:08 +0100 [a6b648645] - Merge pull request #1490 from openWB/master', '2024-03-14 11:48:27 +0100 [1d1d842c5] - Merge pull request #1367 from openWB/master'], 'datastore_version': 38, 'dataprotection_acknowledged': True, 'usage_terms_acknowledged': True, 'debug_level': 10, 'release_train': 'master', 'version': '2.1.3-Patch.1', 'boot_done': True, 'lastlivevaluesJson': {'timestamp': 1711480560, 'time': '20:16:00', 'grid': 0.0, 'house-power': 0.0, 'charging-all': 0.0, 'pv-all': -0.0, 'cp1-power': 0.0}, 'module_update_completed': True}
2024-03-26 20:17:40,576 - {control.data:279} - {INFO:MainThread} - device0
{'name': 'Virtuelles Gerät', 'type': 'virtual', 'id': 0, 'configuration': {}}
2024-03-26 20:17:40,580 - {control.data:281} - {INFO:MainThread} - component0
{'name': 'Virtueller Zähler', 'type': 'counter', 'id': 0, 'configuration': {'external_consumption': 0}}
2024-03-26 20:17:40,584 - {control.data:279} - {INFO:MainThread} - device1
{'name': 'MQTT', 'type': 'mqtt', 'id': 1, 'configuration': {}}
2024-03-26 20:17:40,588 - {control.data:281} - {INFO:MainThread} - component3
{'name': 'MQTT-Wechselrichter', 'type': 'inverter', 'id': 3, 'configuration': {}}
2024-03-26 20:17:40,590 - {control.data:253} - {INFO:MainThread} - 

2024-03-26 20:17:40,597 - {control.algorithm.algorithm:25} - {INFO:MainThread} - # Algorithmus
2024-03-26 20:17:40,602 - {control.algorithm.algorithm:30} - {INFO:MainThread} - **Mindestrom setzen**
2024-03-26 20:17:40,605 - {control.algorithm.algorithm:32} - {INFO:MainThread} - **Sollstrom setzen**
2024-03-26 20:17:40,607 - {control.counter:249} - {DEBUG:MainThread} - Anpassen des Regelbereichs 0W
2024-03-26 20:17:40,611 - {control.counter:217} - {INFO:MainThread} - Überschuss zur PV-geführten Ladung: 0W
2024-03-26 20:17:40,612 - {control.counter:484} - {DEBUG:MainThread} - Zähler 0: Begrenzung der verbleibenden Leistung auf 0W
2024-03-26 20:17:40,613 - {control.algorithm.algorithm:43} - {INFO:MainThread} - **Keine Leistung für PV-geführtes Laden übrig.**
2024-03-26 20:17:40,613 - {control.process:23} - {INFO:MainThread} - # Ladung starten.
2024-03-26 20:17:40,618 - {modules.common.component_context:25} - {DEBUG:set current cp1} - Update Komponente ['Ladepunkt']
2024-03-26 20:17:40,658 - {control.counter:193} - {INFO:MainThread} - 0W reservierte EVU-Leistung, 0W freigegebene EVU-Leistung
2024-03-26 20:17:45,973 - {schedule:660} - {DEBUG:MainThread} - Running job Job(interval=1, unit=minutes, do=smarthome_handler, args=(), kwargs={})
```

benderl, 2024-03-28:
Da Du selbst von einem Eigenbau mit anderem Display schreibst, können wir da nicht helfen. Bitte im Forum besprechen. Dort hast Du ja bereits die Info gepostet.

closed 2024-03-28 by benderl

nicomania, 2024-03-28:
Wow. Eine derart abschmetternde Antwort habe ich nicht erwartet.

Nein, es handelt sich um das gleiche Display, wie bei der Verkaufsversion. Es ist richtig, dass ich dieses Problem schon im Forum gepostet habe, doch dort wird es vollkommen ignoriert. Wahrscheinlich, weil sich dort gefühlt nur Endverbraucher treffen. Ich dachte, dass ein derartiges Problem doch besser bei den Entwicklern aufgehoben ist.

Ebenfalls dachte ich, dass es sich hier um ein „offenes“ Projekt handelt, jedenfalls legt das der Name OpenWB nahe. Lebt dieses Projekt etwa nicht von den Eigenbauern, die den Entwicklern helfen Probleme aufzuspüren und zu beseitigen, die für die Verkaufsversion ebenfalls von Bedeutung sind? Dann habe ich da wohl was falsch verstanden. 

Ich finde es aber auch bemerkenswert, dass Leuten wie mir nicht einmal das Werkzeug (SSH Zugriff) an die Hand gegeben wird, um selbst Probleme zu lösen, von der die Gemeinschaft wiederum profitieren könnte.

Ok, dann bin ich jetzt stuck auf V1.9, obwohl ich dieselbe Hardware benutze. 
Ich danke ganz herzlich für diese Erfahrung. Alles Gute.

benderl, 2024-03-28:
Sorry, wenn das falsch rüber gekommen ist. Ich versuche es nochmal etwas genauer zu erklären.

Du verwendest vermutlich unser bereitgestelltes Image. Das ist auf den Einsatz bei den Produkten aus dem Shop von openWB zugeschnitten. Unterstützung für andere Hardware können wir dort nicht einbauen.

Dein Problem scheint auch nicht wirklich etwas mit der openWB Software zu tun haben, sondern mit dem im Image genutzten Raspberry Pi OS. Du müsstest dann selber ein entsprechendes Basisimage verwenden und kannst die software2 wie auch früher die 1.9 darauf installieren.

Da das Problem nichts mit der software2 aus diesem Repo zu tun hat, sondern dem Image, ist es hier fehl am Platz. Hier werden allgemeine Probleme mit der Regelung, Anbindung von Komponenten etc. gemeldet. Dein Problem ist besser im Forum aufgehoben, da es sehr individuell ist und der Code in diesem Repo nicht die Ursache sein kann.

nicomania, 2024-04-04:
Danke, mit dieser Antwort kann ich mehr anfangen. Ich habe es nun tatsächlich zu Laufen gebracht. Neues Bullseye image aufgesetzt und das Installationsskript aus diesem Repo gestartet. Danke für den Hinweis, ich werde die Info gleich an die anderen Betroffenen aus dem Forum weitergeben.