openWB/core #1626: Wechseln des Entwicklungszweigs liefert Stacktrace

scoopex, 2024-05-18:
Beim Wechseln des Entwicklungszweigs von BETA zu RELEASE erhalte ich folgenden Stracktrace:
(Die Software läuft auf RELEASE Version "2024-05-03 14:13:39 +0200 [bc3399944]"

```
Es ist ein interner Fehler aufgetreten: Traceback (most recent call last): File "/var/www/html/openWB/packages/helpermodules/command.py", line 129, in on_message func(connection_id, payload) File "/var/www/html/openWB/packages/helpermodules/command.py", line 638, in systemUpdate self.createCloudBackup(connection_id, {}) File "/var/www/html/openWB/packages/helpermodules/command.py", line 688, in createCloudBackup SubData.system_data["system"].create_backup_and_send_to_cloud() File "/var/www/html/openWB/packages/helpermodules/system.py", line 97, in create_backup_and_send_to_cloud self.backup_cloud.update(backup_filename, data) File "/var/www/html/openWB/packages/modules/common/configurable_backup_cloud.py", line 15, in update self.__component_updater(backup_filename, backup_file) File "/var/www/html/openWB/packages/modules/backup_clouds/nextcloud/backup_cloud.py", line 36, in updater upload_backup(config.configuration, backup_filename, backup_file) File "/var/www/html/openWB/packages/modules/backup_clouds/nextcloud/backup_cloud.py", line 17, in upload_backup raise ValueError(f"URL '{config.ip_address}' hat nicht die erwartete Form " ValueError: URL 'https://www.redacted.org/nextcloud/index.php/s/NnQjFs4EqqREDACTED' hat nicht die erwartete Form 'https://server/index.php/s/user_token' oder 'https://server/s/user_token' 
```
Schalte ich automatische Backups für Updates ab, tritt das Problem nicht mehr auf.

benderl, 2024-05-18:
Für individuelle Probleme bitte das Forum nutzen. Hier sollten nur allgemeingültige Probleme mit der Software als Issue eingereicht werden. Kann gut sein, dass Dein Problem im Forum schnell behoben werden kann. Wenn sich dann herausstellt, dass es ein generelles Problem ist, gerne wieder hier melden.

scoopex, 2024-05-18:
@benderl Ich denke das ist kein individuelles Problem. Ich vermute das Problem lässt sich durch eine nicht valide Nextcloud URL für die Backup Location provozieren.

scoopex, 2024-05-21:
Ah interessant. Ich hatte "NnQjFs4EqqREDACTED" als das User Token verstanden :-) 

Beim Einrichten der Nextcloud Anbindung hat das keinen Fehler erzeugt/gezeigt!
Das müsste auch dort geprüft und entsprechend moniert werden.
Ich habe mich auch gewundert warum der Upload von Backups nie funktioniert hat, aber die Einstellungen belassen, weil ich vermutet hatte das dieser "Bug" später gefixed wird....

Unterstützt OpenWB den Upload auf ein Shared Folder, den ein Nutzer auf einen bestimmten Ordner angelegt hat, nicht?
Falls nicht: Warum nicht, das wäre doch das naheliegendste Szenario?
Die allgemeinen Zugangsdaten würde ich schon aus Security-Gründen nie in OpenWB eintragen weil mir das zu heikel wäre. Selbst wenn OpenWB 2.0 die lang ersehnte Nutzeranmeldung schon mitbringen würde.

pama87, 2024-05-22:
> Ah interessant. Ich hatte "NnQjFs4EqqREDACTED" als das User Token verstanden :-)
>
> Beim Einrichten der Nextcloud Anbindung hat das keinen Fehler erzeugt/gezeigt! Das müsste auch dort geprüft und [...]

Hallo scoopex,
kurze Rückfrage: Ist das eine gültige Nextcloud URL (https://www.redacted.org/nextcloud/index.php/s/NnQjFs4EqqREDACTED)?

scoopex, 2024-05-22:
Ich habe den Link mit dem Hinzufügen von "REDACTED" aus Sicherheitsgründen etwas verfälscht.
Anstatt von REDACTED stehen andere Zeichen.

https://www.256bit.org/nextcloud/index.php/s/NnQjFs4Eqq6TSfQ

pama87, 2024-05-23:
Ah, dann ist https://www.256bit.org/nextcloud eine gültige URL, die vom parser bisher noch nicht in der Form erkannt wird

pama87, 2024-05-23:
> Ich habe den Link mit dem Hinzufügen von "REDACTED" aus Sicherheitsgründen etwas verfälscht. Anstatt von REDACTED stehen andere Zeichen.
> 
> Ein vollständiges Beispiel: https://www.256bit.org/nextcloud/index.php/s/NnQjFs4Eqq6TSfQ

Ich kann zwar die URL in den parser mit aufnehmen, jedoch kann ich nicht prüfen, ob der upload link korrekt ist. Daher die Frage: Ist dieser Link hier korrekt: https://www.256bit.org/public.php/webdav/openWB_backup_2024-05-23_10-13-04.tar.gz oder muss der link dann https://www.256bit.org/nextcloud/public.php/webdav/openWB_backup_2024-05-23_10-13-04.tar.gz heissen?

pama87, 2024-05-23:
Der Teil hinter webdav/ ist der dann generierte filename

scoopex, 2024-05-23:
Das kann ich nicht sagen.
Gerne kann  ich Ihnen einen Testshare für Nextcloud zur Verfügung stellen, falls sie kein eigens haben.
(Mail an ms-github@256bit.org)

pama87, 2024-05-23:
> Das kann ich nicht sagen. Gerne kann ich Ihnen einen Testshare für Nextcloud zur Verfügung stellen, falls sie kein eigens haben. (Mail an [ms-github@256bit.org](mailto:ms-github@256bit.org))

Danke. Falls es nicht funktioniert, dann bräuchte ich einen Testaccount. Ich habe es soweit gefixt:
1. Das Parsen
2. Die Fehlerbehandlung
Sobald es in den Master gemerged wurde, bitte testen und falls es nicht geht bitte um Rückmeldung
Danke für die Unterstützung

closed 2024-06-18 by LKuemmel