openWB/core #1636: Verbindung zu Victron 3P75CT Stromzähler schlägt fehl

AlexHuebi, 2024-05-26:
Hallo, 
ich wollte heute meinen Victron Stromzähler zur OpenWB Software hinzufügen, bekomme aber nur die Fehlermeldung: 

```
<class 'pymodbus.exceptions.ConnectionException'> ('Failed to connect[ModbusTcpClient(192.168.0.41:502)]', 'Modbus-Client konnte keine Verbindung zu 192.168.0.41:502 aufbauen. Bitte Einstellungen wie IP-Adresse, Ladepunkt-Typ, .. und Hardware-Anschluss prüfen.')
```
Ich habe auch schon eine Vermutung warum dieser Fehler auftritt: 
Der Stromzähler im Standalone Modus unterstützt kein Modbus/TCP sondern nur Modbus/UDP. 

Also müsste hierfür die im Hintergrund verwendete Bibliothek angepasst werden. 

Außerdem sind die Register, welche der Stromzähler zur Verfügung stellt, unterschiedlich zu den Registern welche über die GX Gerätereihe unterschiedlich sind.

benderl, 2024-06-03:
Für den Zähler müsste ein komplett neues Modul geschrieben werden, das kann aktuell noch nicht funktionieren.

Kannst Du bitte mehr Infos zu dem Zähler bereitstellen? Falls wirklich Modbus/TCP verwendet werden muss, sollte es vom Hersteller eine detaillierte Info zu den verfügbaren Registern geben.

AlexHuebi, 2024-06-03:
Ich habe schon überall gesucht, nur leider konnte ich keine offizielle  Liste der Register oder jegliche andere Dokumentation zu dem Modbus Interface gefunden. Ich hatte mich aber mit einem Modbus Client von meinem PC aus verbunden, welcher die Verbindung zu dem Stromzähler verweigerte wenn dieser im TCP Modus war. Sobald ich auf UDP umgestellt hatte, konnte ich auch die Werte auslesen.  

Die einzige Referenz aus "offizieller Quelle" konnte ich nur in einem Pythonmodul von VE finden. https://github.com/victronenergy/dbus-modbus-client/blob/master/victron_em.py 

Falls es hilft, habe ich aus den Victron Foren eine Homeassistant Modbus config gefunden, welche auch bei mir funktioniert. https://community.victronenergy.com/questions/258646/modbus-id-and-register-address-of-the-vm-3p75ct.html

<details>
  <summary>Die Config sieht wie hier aus:</summary>

```yml
modbus:
  - name: "VM-3P75CT"
    type: udp
    host: 192.168.0.41
    port: 502
    sensors:
      - name: 'VM-3P75CT Power'
        unique_id: vm_3p75ct_power
        unit_of_measurement: W
        slave: 1
        address: 0x3080
        data_type: int32
        scan_interval: 1
        device_class: power
      - name: 'VM-3P75CT Forward'
        unique_id: vm_3p75ct_forward
        unit_of_measurement: kWh
        slave: 1
        address: 0x3034
        data_type: uint32
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: energy
        state_class: total_increasing
      - name: 'VM-3P75CT Reverse'
        unique_id: vm_3p75ct_reverse
        unit_of_measurement: kWh
        slave: 1
        address: 0x3036
        data_type: uint32
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: energy
        state_class: total_increasing
      - name: 'VM-3P75CT PENVoltage'
        unique_id: vm_3p75ct_penvoltage
        unit_of_measurement: V
        slave: 1
        address: 0x3033
        data_type: int16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: voltage
      - name: 'VM-3P75CT Frequency'
        unique_id: vm_3p75ct_frequency
        unit_of_measurement: Hz
        slave: 1
        address: 0x3032
        data_type: uint16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: frequency

      - name: 'VM-3P75CT Voltage L1'
        unique_id: vm_3p75ct_voltage_l1
        unit_of_measurement: V
        slave: 1
        address: 0x3040
        data_type: int16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: voltage
      - name: 'VM-3P75CT Current L1'
        unique_id: vm_3p75ct_current_l1
        unit_of_measurement: A
        slave: 1
        address: 0x3041
        data_type: int16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: current
        
      - name: 'VM-3P75CT Voltage L2'
        unique_id: vm_3p75ct_voltage_l2
        unit_of_measurement: V
        slave: 1
        address: 0x3048
        data_type: int16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: voltage
      - name: 'VM-3P75CT Current L2'
        unique_id: vm_3p75ct_current_l2
        unit_of_measurement: A
        slave: 1
        address: 0x3049
        data_type: int16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: current

      - name: 'VM-3P75CT Voltage L3'
        unique_id: vm_3p75ct_voltage_l3
        unit_of_measurement: V
        slave: 1
        address: 0x3050
        data_type: int16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: voltage
      - name: 'VM-3P75CT Current L3'
        unique_id: vm_3p75ct_current_l3
        unit_of_measurement: A
        slave: 1
        address: 0x3051
        data_type: int16
        scale: 0.01
        precision: 2
        scan_interval: 1
        device_class: current
```
</details>

AlexHuebi, 2026-03-04:
wurde in #3143 implementiert