openWB/core #1734: backup-cloud samba Konfiguration mit leerem String für Unterordner (optional)führt zu python error

Carstig, 2024-07-07:
Beim Einrichten des Backup mit Samba, habe ich das Feld für 'Unterordner (optional)' leer gelassen.

Beim Start von "Manuelle Cloud-Sicherung" bekam ich dann folgenden Fehler:
```
2024-07-07 10:09:08,614 - {SMB.SMBConnection:104} - {INFO:Commands} - Authentication with remote machine "192.168.0.99" for user "openwb" will be using NTLM v2 authentication (with extended security)
2024-07-07 10:09:08,616 - {helpermodules.messaging:42} - {DEBUG:Commands} - pub_message: message: 'Es ist ein interner Fehler aufgetreten: Traceback (most recent call last):
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 129, in on_message
    func(connection_id, payload)
  File "/var/www/html/openWB/packages/helpermodules/command.py", line 700, in createCloudBackup
    SubData.system_data["system"].create_backup_and_send_to_cloud()
  File "/var/www/html/openWB/packages/helpermodules/system.py", line 97, in create_backup_and_send_to_cloud
    self.backup_cloud.update(backup_filename, data)
  File "/var/www/html/openWB/packages/modules/common/configurable_backup_cloud.py", line 15, in update
    self.__component_updater(backup_filename, backup_file)
  File "/var/www/html/openWB/packages/modules/backup_clouds/samba/backup_cloud.py", line 59, in updater
    upload_backup(config.configuration, backup_filename, backup_file)
  File "/var/www/html/openWB/packages/modules/backup_clouds/samba/backup_cloud.py", line 31, in upload_backup
    found_invalid_chars = re.search(r'[\\\:\*\?\"\<\>\|]+', config.smb_path)
  File "/usr/lib/python3.9/re.py", line 201, in search
    return _compile(pattern, flags).search(string)
TypeError: expected string or bytes-like object
' type: 'MessageType.ERROR' target: 'MessageTarget.USER'
2024-07-07 10:09:08,617 - {helpermodules.messaging:60} - {ERROR:Commands} - Messaging: Fehlermeldung: {'source': 'command', 'type': 'danger', 'message': 'Es ist ein interner Fehler aufgetreten: Traceback (most recent call last):\n  File "/var/www/html/openWB/packages/helpermodules/command.py", line 129, in on_message\n    func(connection_id, payload)\n  File "/var/www/html/openWB/packages/helpermodules/command.py", line 700, in createCloudBackup\n    SubData.system_data["system"].create_backup_and_send_to_cloud()\n  File "/var/www/html/openWB/packages/helpermodules/system.py", line 97, in create_backup_and_send_to_cloud\n    self.backup_cloud.update(backup_filename, data)\n  File "/var/www/html/openWB/packages/modules/common/configurable_backup_cloud.py", line 15, in update\n    self.__component_updater(backup_filename, backup_file)\n  File "/var/www/html/openWB/packages/modules/backup_clouds/samba/backup_cloud.py", line 59, in updater\n    upload_backup(config.configuration, backup_filename, backup_file)\n  File "/var/www/html/openWB/packages/modules/backup_clouds/samba/backup_cloud.py", line 31, in upload_backup\n    found_invalid_chars = re.search(r\'[\\\\\\:\\*\\?\\"\\<\\>\\|]+\', config.smb_path)\n  File "/usr/lib/python3.9/re.py", line 
```

Erst nachdem ich einen Unterordner auf dem samba server erstellt habe und das Feld mit 'openwb/' gefüllt habe hat es geklappt.

openWB version `2.1.4`  `2024-06-04 08:52:24 +0200 [417a30647]` 

möglicher fix in : `packages/modules/backup_clouds/samba/config.py`  Zeile 5. Default Wert für `smb_path` sollte `/`  statt `None` 

habe ich aber nicht getestet (und ehrlicherweise sieht das mehr nach patch aus). Eher der nutzende Code sollte abprüfen. Wenn der default allerdings auch gleich in der GUI sichtbar wird, wäre meine Änderung allerdings auch nicht schlecht.

pama87, 2024-07-08:
Hallo Carstig,
hast du das wiki gelesen (https://github.com/openWB/core/wiki/Samba-als-Sicherung-einrichten)?
Darin steht eine detaillierte Anleitung für das Einrichten von Samba. Darin steht auch, dass 'openwb/'  bei Unterordner nicht leer sein darf (Schritt 13 der Anleitung z.B. test/).Das Wort optional ist leider irreführend.

pama87, 2024-07-08:
> Beim Einrichten des Backup mit Samba, habe ich das Feld für 'Unterordner (optional)' leer gelassen.
> 
> Beim Start von "Manuelle Cloud-Sicherung" bekam ich dann folgenden Fehler:
> 
> 
> möglicher fix in : `packages/modules/backup_clouds/samba/config.py` Zeile 5. Default Wert für `smb_path` sollte `/` statt `None`
> 
> habe ich aber nicht getestet (und ehrlicherweise sieht das mehr nach patch aus). Eher der nutzende Code sollte abprüfen. Wenn der default allerdings auch gleich in der GUI sichtbar wird, wäre meine Änderung allerdings auch nicht schlecht.

Hallo Carstig,
hast du das wiki gelesen (https://github.com/openWB/core/wiki/Samba-als-Sicherung-einrichten)?
Darin steht eine detaillierte Anleitung für das Einrichten von Samba. Darin steht auch, dass 'openwb/' bei Unterordner nicht leer sein darf (Schritt 13 der Anleitung z.B. test/).Das Wort optional ist leider irreführend.

Carstig, 2024-07-09:
Moin, das Wiki habe ich dann auch gesehen. Aber auch hier steht als Beispiel eben ein Unterordner. Und ich hatte das so bei mir nicht vorgesehen, so daß ich den Unterordner leer gelassen habe. 

Das ganze ist also weniger ein funktionales sondern ein Benutzerführungsproblem. Dann wäre der Fix eher in der Gui und das "optional" zu streichen. 

Sobald wir uns geeinigt haben, kann ich einen fix versuchen (bin des Python mächtig)

pama87, 2024-07-11:
> Moin, das Wiki habe ich dann auch gesehen. Aber auch hier steht als Beispiel eben ein Unterordner. Und ich hatte das so bei mir nicht vorgesehen, so daß ich den Unterordner leer gelassen habe.
>
> Das [...]

ok. Mach gerne einen PR und ich seh mir das an und prüfe es dann bei mir lokal nach. Danke

closed 2024-07-11 by benderl

Carstig, 2024-07-11:
Du bist der Hammer! Besser als ich es je avisiert hätte.