openWB/core #1904: bat_all.py _max_bat_power_hybrid_system

ChrisMuki, 2024-09-29:
This method has changed its calculation and is now wrong!
OLD: max_bat_discharge_power = parent_data.config.max_ac_out + \
                    parent_data.get.power + max(battery.data.get.power, 0)
NEW: max_bat_discharge_power = parent_data.config.max_ac_out + \
                    parent_data.get.power + min(battery.data.get.power, 0)

ChrisMuki, 2024-09-29:
Correct would be: 
max_bat_discharge_power = parent_data.config.max_ac_out +
parent_data.get.power +battery.data.get.power

Examples see in attached Screenshot

[image]

LKuemmel, 2024-09-30:
Die Tabellen sind gut!

Bei der letzten Tabelle das erste Ergebnis kann ich nicht nachvollziehen. Der WR speist 8kW ins Haus ein, der Speicher lädt mit 5 kW, dh es werden 13kW erzeugt. Davon können max 10kW abgegeben werden, dh der Speicher kann mit max 2 kW entladen werden, dann ist die Grenze des WR erreicht. 3 kW müssen immer in den Speicher geladen werden, da die Ausangsleistung des WR diese nicht ausgegeben kann.
Wie kommst Du auf 7kW?

ChrisMuki, 2024-09-30:
Der Wert parent.data.get.power ist zumindest laut meinen Beobachtungen die gesamte PV Leistung, nicht nur der AC output. Sonst hätte ich bei meinem WR ja keine -12000 beobachten können (der kann nur 10kW AC)

Daher ist zu diesem Zeitpunkt

- die gesamte PV Leistung -8000
- die Batterie lädt mit 5000
- => kann also maximal 2000W entladen
- => es stehen maximal noch 7000W Leitung durch die Batterie zur Verfügung.

Ich habe gerade gesehen, dass die Test nicht laufen.....

ChrisMuki, 2024-09-30:
@LKuemmel Wenn du bestätigen kannst, dass parent.data.get.power
im Fall eines Hybrid-WR nicht die AC Leistung sondern die PV Leistung sein soll (abgesehen von Wandlungsverlusten....) dann würde ich die Tests auch dahingehen anpassen um die Fälle wie von mir dargestellt abzudecken.

ChrisMuki, 2024-10-06:
Habe die Tests entsprechend angepasst und alle möglichen Fälle abgedeckt :)

LKuemmel, 2024-10-11:
Danke für deine ausführlichen Überlegungen!

closed 2024-10-11 by LKuemmel